Fornoth/spotify-connect-web #10: cannot load libspotify_embedded_shared.so

insanebits, 2015-09-17:
Running main.py results in missing spotify_embedded_shared.so library. Exact output:

python main.py 
Traceback (most recent call last):
  File "main.py", line 8, in <module>
    from connect_ffi import ffi, lib
  File "/home/pi/spotify/spotify-connect-web/connect_ffi.py", line 19, in <module>
    libraries=[str('spotify_embedded_shared')])
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 405, in verify
    lib = self.verifier.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 101, in load_library
    return self._load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 211, in _load_library
    return self._vengine.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 155, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/home/pi/spotify/spotify-connect-web/_cffi__x8b237976x2c265354.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory

I can confirm that library is in the same directory as main.py and  I also tried symlinking to /usr/lib and /usr/local/lib which didn't change anything.

I am running on Raspberry Pi2, and strange enough given docker image works as it's supposed. On the docker image libspotify_embedded_shared.so is placed in /usr/lib directory which I already tried without any luck.

Any suggestions how to get it working without docker?

balbuze, 2015-10-09:
I've got the same problem with RPI B+ and RPI2
With the chroot version, 
And has anyone managed to use a software mixer with it ?

Fornoth, 2015-10-15:
@insanebits make sure you're using the armhf version of the library if you're using it outside of an chroot
@balbuze rpi b+ requires the chroot/docker version, and to use it without chroot/docker on rpi 2, you need to use the armhf version, and software mixer stuff is being discussed in #11

balbuze, 2015-10-15:
Hello,
thank a lot for the answers.
On my RPI2 I, to use it without chroot or docker, I have put  libspotify_embedded_shared.so (armhf version) in same folder than python script.
Now I've got a other message 

.//libspotify_embedded_shared.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "main.py", line 8, in <module>
    from connect_ffi import ffi, lib
  File "/root/connect/connect_ffi.py", line 19, in <module>
    libraries=[str('spotify_embedded_shared')])
  File "/usr/lib/python2.7/dist-packages/cffi/api.py", line 340, in verify
    lib = self.verifier.load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 74, in load_library
    self._compile_module()
  File "/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 139, in _compile_module
    outputfilename = ffiplatform.compile(tmpdir, self.get_extension())
  File "/usr/lib/python2.7/dist-packages/cffi/ffiplatform.py", line 25, in compile
    outputfilename = _build(tmpdir, ext)
  File "/usr/lib/python2.7/dist-packages/cffi/ffiplatform.py", line 51, in _build
    raise VerificationError('%s: %s' % (e.__class__.__name__, e))
cffi.ffiplatform.VerificationError: LinkError: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

Any idea ?

maumi, 2015-10-15:
Look in #11, answer is in my post.

Fornoth, 2016-03-29:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-03-29 by Fornoth

RafaPolit, 2017-01-20:
This was important stuff! Why isn't this on the readme?  Please add the fact that the armhf version of the lib is needed, it would be much clearer!

Thanks,
Rafa.