Fornoth/spotify-connect-web #11: Software Volume Mixing

artistro08, 2015-10-01:
this is more of a question. will software side volume be used in future releases? Just asking. My current DAC doesn't support alsamixer's gui. so this would be dope if you could

Fornoth, 2015-10-06:
You can try using http://alsa.opensrc.org/How_to_use_softvol_to_control_the_master_volume for a virtual mixer in alsa

maumi, 2015-10-06:
softvol works in bash (speaker-test) but in pyalsaaudio doesn't seem to work. Tried to change volume with webpage and with alsamixer. Strangely volume is changed in alsamixer when changed in webpage. Can't use softvol device with -D. Using -m also doesn't work.

DieterKoblenz, 2015-10-07:
Using the spotify app I can control the volume of the DAC. However it goes silent at about 80% or so. Is there a way to tune this so that you can control the volume better?

maumi, 2015-10-07:
so is the volume quiet or really silent at 80%? Maybe first you can try to compare by opening alsamixer in shell and then change volume in app.
If the levels are identical, you could change the source code to your needs, but sounds strange that it's silent at 80%.

Fornoth, 2015-10-09:
@maumi if you have a rpi2 (or armv7 device) you can use the hardfloat libspotify_embedded, rpi1 you have to use a chroot or docker
@DieterKoblenz it's quiet at 0%-80%?

DieterKoblenz, 2015-10-09:
@Fornoth yes, from 0 - ~80 its empty. In alsa I can change the volume without a problem to the level I want but on the spotify app this is difficult because I have to work in the 20% range.

DieterKoblenz, 2015-10-09:
I've checked again and if I turn up the volume on my amp it is not completely silent at 80, it stops at 51 or so, and then it is completely silent.

maumi, 2015-10-11:
Habe you tried to open alsamixer in shell and then change volume in spotify APP or web Browser to 50% and look if it's also at 50%? If it's at 0% it should be a scaling Problem in your case and easy fixable in code for you.

artistro08, 2015-10-12:
I tried that. for some reason Connect is changing the vol on the default RPi card. I keep specifying the device to use and it keeps controlling the volume on the default card. If i do "speaker-test -Dsoftvol -c 2 -twav" it  plays on the right mixer.

maumi, 2015-10-12:
@DieterKoblenz so your volume problem is solved?
@artistro08 have you tried to overwrite the default device with softvol?

DieterKoblenz, 2015-10-12:
No, I'm still trying to find a way to change the way spotify handles the volume. To make it clear, I would like spotify's (0-100) scale to control alsa's (50-100) volume.

Not sure if thats even possible

maumi, 2015-10-12:
you could try to change line 151 in console_callbacks.py
it is now: mixer.setvolume(int(volume / 655.35))
to: mixer.setvolume(int(50 + (volume / 655.35) / 2 ))

untested, but should do what you want to achieve.

artistro08, 2015-10-12:
@maumi Yes I did. I'm thinking if i disabled the onboard sound card it'll work

artistro08, 2015-10-13:
here's what happened when I used softvol:
`
root@Maiku:/app# cd /app && python main.py --username artistro08 --password twilight17 --name Maiku --bitrate 320 
Traceback (most recent call last):
  File "main.py", line 9, in <module>
    from connect import Connect
  File "/app/connect.py", line 8, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/app/console_callbacks.py", line 16, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
alsaaudio.ALSAAudioError: No such file or directory
`

maumi, 2015-10-13:
I'm really no expert with alsa but seems that your device has no mixers. Have you tried speaker-test with default device and changing volume while running? 
After a reboot it seems that the file with softvol js only loaded when you use the alsa device, so you need to start speaker-test once before starting spotify. You can easily see if mixer is loaded by calling alsamixer and look if mixers are present. 
Another thing: do you have a pi2 and get it running without chroot and docker?
If yes, could you help me a little with that

artistro08, 2015-10-14:
I do have a Pi2. also, the device did have a mixer enabled. it wouldn't recognize that. As far as getting it running without chroot and docker, I upgraded to Jessie and installed the libraries i needed i to run via pip. then I put the libspotify-embedded-shared.so in the libs directory and ran accordingly

maumi, 2015-10-14:
thanks, got it working :-) I took the wrong library (Rocki). Now I finally have softvol with spotify-connect. 
If you are interested in what I did:
my .asoundrc is very simple, 
pcm.softvol {
    type            softvol
    slave {
        pcm         "default"
    }
    control {
        name        "Master"
        card        0
    }
}

So I don't override the default sound output.
Next important step is to get the correct version of alsaaudio.
you need to get a version from up to 0.8, so 
pip install pyalsaaudio>=0.8
Otherwise the argument device in PCM won't work.
Finally, you need to change the line in console_callbacks.py from
card = args.device)
to
device = args.device).
When you now start your main.py (only get it working with this, not with docker or chroot) with -d softvol it should work.
Hope could help someone.

artistro08, 2015-10-14:
You sir are a genius! got it working with a little more advanced config since i disabled the on-board sound card.

/etc/asound.conf: 
```
pcm.!default {
    type             plug
    slave.pcm       "softvol"
}

pcm.softvol {
    type            softvol
    slave {
        pcm         "plughw:1"
    }
    control {
        name        "Master"
        card        1
    }
}

ctl.!default {
    type            hw
    card            1
}
```
works like a charm! gonna run this along with shairport :)

maumi, 2015-10-14:
thanks ;) glad to hear it's working. maybe you could help me out a little. I also use shairport-sync and spotify together. But because spotify steals the audio device I kill spotify when shairport wants to play.
Before I could easily do it with docker (docker start/stop) but now I tried to run spotify with daemontools. But I couldn't kill it this way.
So quick and dirty for now: start it with script and kill it with 
sudo killall -TERM python
But I know it's far away from a good way to handle it.

artistro08, 2015-10-14:
I've actually figured out a workaround for it. there's a session control that allows you to run commands before and after shairport-sync starts. it should be in the config file under /etc/shairport-sync.conf.
i have scripts setup to kill spotify-connect when airplay is running and start it after its done.

maumi, 2015-10-14:
I've doing the same. Got it working now.
I think issue can be closed now.

closed 2015-10-14 by artistro08

balbuze, 2015-10-14:
@artistro08 
I would like to run it without chroot or docker
When I launch main.py, I still have got a error ending by "  File "/usr/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 149, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/root/connect/__pycache__/_cffi__x8b237976x2c265354.arm-linux-gnueabihf.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory"
Which file do you use to make it work? 
Thank you!

maumi, 2015-10-14:
@balbuze
Should only work on pi2. 
You need to copy libspotify from armhf (spotify Ressources) to /usr/lib/ 
And you need to CD into The folder with main.py

artistro08, 2015-10-14:
What he said. its just dependencies is the main reason why it wont run right off the back.

Fornoth, 2015-10-15:
I'll add an option soon for volume scaling

artistro08, 2015-10-15:
awesome! can't wait.

maumi, 2015-10-15:
Maybe you also could change code like in my comment. But this will break running Version with updated code. Needs to rerun pip then with alsaaudio>=0.8 but only with this change you can select soft devices.

cdsaenz, 2016-03-06:
This is a great conversation, helped me a lot to set my RPI1B+ with both Shairport and Spotify connect (even killing the process when the other takes over). However I cannot get the softvol device to be operational.. Have you got a blog or some detailed instructions to see what I can be missing? Thanks!

maumi, 2016-03-06:
You mean you couldn't change the volume with spotify?

cdsaenz, 2016-03-06:
Exactly @maumi.  By the way the -D parameter didn't stick (if I add it I get an error). Everything else is in there, I think. I am running on a Hifiberry DAC+ output, but I set it as 0. I see the playback_volume (event?) is reported back to the process but there is no change in the output. As I'm using an old remoteless amp that would be a blessing to manage volume via software.

maumi, 2016-03-06:
Afaik hifiberry dac has no softvol and only hardware volume.

cdsaenz, 2016-03-06:
@maumi you gave me a great tip. The instructions at https://support.hifiberry.com/hc/en-us/articles/205377202-Adding-software-volume-control are meant for softvol with the Hifiberry.. I tried them and speaker-test recognizes the "hifiberry" device but I can find no way to active it in spotify-connect-web.. Seems like I'm almost there..

cdsaenz, 2016-03-06:
The problem seems to be that a change is necessary in mpd.conf for this to work.. And afaik spotify- connect-web does not use mpd at all. So dead end for now.

maumi, 2016-03-06:
Should work when it's working with speaker test and if you could change volume in alsamixer.
Take a look in other issues like #37 to see how to set hifiberry in spotify.

cdsaenz, 2016-03-06:
That was another good tip, but I think the issue is .. That I cannot get to change the code below. If I change `card` for `device` (what does make sense as the softvol setting is not really a CARD), the process will fail out immediately. As reported, the new device works ok in speakertest but couldn't find it with F6 in alsamixer. Maybe related?

```
device = alsa.PCM(
        alsa.PCM_PLAYBACK,
        card = args.device)
```

maumi, 2016-03-06:
You need to read the issue 37 in detail. You need a newer version of alsa! 0.8
Have you tried 
alsamixer -c1
?

cdsaenz, 2016-03-06:
I read it and the only things I found different were the ones below. Thanks for your help and sorry for my bad linux! 
- I found no cons_callb.py file in the installation
- to upgrade alsa I had to use: sudo pip3 install pyalsaaudio>=0.8 (it apparently upgraded ok)
- I tried also: sudo apt-get --only-upgrade install python-alsaaudio but it says it's at 0.7.1 and no newer version exist (on armhf)
Maybe I have to upgrade some other alsa package?

maumi, 2016-03-07:
I meant console_callbacks.py
Why do you use python3? Why not python2? Sudo pip install instead of pip3

cdsaenz, 2016-03-07:
I see. At some point the `pip` command didn't work and I must have installed the python3 version. Anyway, I could easily uninstall python3 version of pyalssaudio (0.8.2), just in case. 

Trying `sudo pip install pyalsaaudio>=0.8` didn't work. What did work was `pip install --upgrade pyalsaaudio`. Now I'm on 0.8.2 for python2 afaik (`pip list` says so). 

- As before, `speaker-test -Dhifiberry` works ok.
- `aplay -L` shows hifiberry as a device with no details
- But, `alsamixer -Dhifiberry` won't work, it indicates "invalid CTL"
- Still, couldn't change the line in _cons_callb.py _ as per the instructions. I get the error:

```
pi@raspberrypi:~ $ /home/pi/spotify-connect-web.sh --name rpiConnect1 --username XXX --password XXXX --bitrate 320 --device hifiberry
Traceback (most recent call last):
  File "main.py", line 9, in <module>
    from connect import Connect
  File "/usr/src/app/connect.py", line 8, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/usr/src/app/console_callbacks.py", line 21, in <module>
    device = args.device)
TypeError: 'device' is an invalid keyword argument for this function

```

maumi, 2016-03-07:
I'm no expert with chroot. But I think it's sth. like a pre-build system. So the version of pyalsaaudio is taken, which was used when built.
Try to use docker, but you then need to change requirements.txt 
pyalsaaudio>=0.8

cdsaenz, 2016-03-07:
I see what you mean @maumi I'll either reinstall this portion or the whole SD card. Thanks a lot.

blubhimself, 2016-10-04:
Hey Guys,
I read this thread and it helped me out configuring my RPiZero with spotify-connect-web and pHAT DAC. For those who cant use softvol (from maumi's post) ... try copying your asound.conf to the APP directory (chroot-folder/usr/src/app) of spotify-connect-web-chroot. Then starting it with --playback_device softvol works like charm for me.

tatoosh, 2016-10-17:
@blubhimself i have the same hardware but i always get the error `ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM softvol`

Please post your asound.conf thx!

maumi, 2016-10-18:
Are you using The chroot version, habe copied asound in chroot and --playback-device as Parameter?

blubhimself, 2016-10-19:
@tatoosh my asound.conf looks like this

pcm.softvol {
type softvol
slave {
pcm "default"
}
control {
name "Master"
card 0
}
}

Than copy asound.conf in your spotify-connect-web-chroot/etc folder and start the server using --playback_device softvol

tatoosh, 2016-10-19:
> copy asound.conf in your spotify-connect-web-chroot/etc 

Thx @blubhimself - it worked now, you helped me a lot.

balbuze, 2016-10-21:
I wrote  a plugin for Volumio2 (https://volumio.org) that installs in one click spotify connect -web on RPI1, RP2 or RPI3 and Sparky (and probably other armv6 and armv7 devices). It use standard version or chroot version depending your device through a detection. You just have to enter your credentials. It works now pretty well. But in the chroot version with software mixer I get no sound... I followed this thread, but I can't make it work. When I change volume in Spotify, I see the volume changing in alsamixer... Any idea ? If you want to test the plugin : https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect

RafaPolit, 2017-01-24:
@Fornoth Has this been finally implemented or are we still to follow these procedures to get software volume? Thanks,

Rafa.

Fornoth, 2018-01-15:
@RafaPolit sorry for the really late reply, but this is something you'll have to do at the ALSA level, because I have no plans of implementing it. Although if somebody send in a pull request for it, I would be fine with that