Fornoth/spotify-connect-web #12: Error when waiting a long time before playing a song

MagicMicky, 2015-10-04:
I set up spotify connect web using the chroot technique on my raspberry pi (B+). When I stop playing songs around mignight, and play something the day after, it usually won't work. The track seems to buffer forever (on the app you hit play, the timer starts, but if you hit pause then play, it starts back to 0).

I activated the logs to check what was going on and it seems that there sometime is an error with ALSA.

```
Fri, 02 Oct 2015 07:38:45 +0000 ESDException in thread Thread-1:
Fri, 02 Oct 2015 07:38:45 +0000 Traceback (most recent call last):
Fri, 02 Oct 2015 07:38:45 +0000 File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
Fri, 02 Oct 2015 07:38:45 +0000 self.run()
Fri, 02 Oct 2015 07:38:45 +0000 File "/usr/lib/python2.7/threading.py", line 763, in run
Fri, 02 Oct 2015 07:38:45 +0000 self.__target(*self.__args, **self.__kwargs)
Fri, 02 Oct 2015 07:38:45 +0000 File "/usr/src/app/console_callbacks.py", line 110, in playback_thread
Fri, 02 Oct 2015 07:38:45 +0000 device.write(data)
Fri, 02 Oct 2015 07:38:45 +0000 ALSAAudioError: Input/output error
```

Not sure where this would come from. [here](http://pastebin.com/9cGhH4wR) is a full detailed log of over 2 days of my raspberry pi running and the problem is shown on line 3434.

Fornoth, 2016-01-08:
Try with the latest version and see if it helps

AlexGustafsson, 2016-05-07:
Same issue here, running latest Chroot. I will check my logs and see if there's anything I can add.