Fornoth/spotify-connect-web #14: Illegal instruction

artistro08, 2015-10-15:
I cloned my original installation of the pi and i get the error "Illegal instruction" on the new pi. everything is set up accordingly. just don't know why its not running. doesn't give me any output at all

Fornoth, 2015-10-15:
Which library are you using, and are you using a pi 1 or pi 2?

artistro08, 2015-10-15:
Pi B+. i cloned it from a Pi2 with some modifications

artistro08, 2015-10-15:
and the armfh version

Fornoth, 2015-10-15:
You'll have to use the chroot/docker version with the rocki library if you're not on rpi 2 (or an armv7+ device, rpi 1 is armv6)

artistro08, 2015-10-15:
ahh. thought so. thank ya.

closed 2015-10-15 by artistro08

deify, 2017-02-09:
I tried to get spotify connect running on my pi b+ without using the chroot because i wanted to add some functionallity to use cec-client commands in pyhton.

is there any way to get it working on the pi b+? I didnt manage to get the libcec working inside the chroot, neither did i manage to run it from source.

Fornoth, 2018-01-15:
Really the only way to get it working on the B+ right now without a chroot is to use a softfloat OS. I don't have any recommendations on one though.