Fornoth/spotify-connect-web #15: TODO List

Fornoth, 2015-10-15:
- [ ] Add volume scaling option (#11)
- [ ] Add option to close/open ALSA device/restart audio thread (#11)
- [ ] Give more information on library not found errors

celeryclub, 2016-04-11:
@Fornoth What's blocking the upgrade to pyalsaaudio 0.8+? I can upgrade on my Pi3 with no problems. Is there particular hardware that it doesn't currently work on?

Fornoth, 2016-04-13:
@celeryclub pyalsaaudio 0.8+ itself works fine, the note is more to upgrade to the newer device/mixer selection features in it

celeryclub, 2016-05-04:
@Fornoth Understood. Thanks.

makruiten, 2016-11-25:
> Handle when a device doesn't have a mixer
I guess this is the reason why my HifiBerry DAC light won't play along with Spotify Connect Web. Is this something that's very hard to fix? If not, can you point me in the right direction, so I can see if I can figure it out?

awiouy, 2016-11-25:
@makruiten 
I use the following for my HifiBerry DAC+ on libreELEC:
playback device: default:CARD=sndrpihifiberry
mixer: Digital

makruiten, 2016-11-26:
@awiouy 
Is that the light version? The other two have mixers, but the light version lacks one.

ClemaX, 2016-11-30:
@makruiten For me the DAC+ Light works, I have got no volume control tho. I'm using the latest runeaudio for the raspberry pi 3.

makruiten, 2016-12-01:
@ClemaX It was probably due to the way it is packaged in Kodi. I had no problems getting it to work in Volumio. In Kodi you are forced to select a mixer, while there are none.

ClemaX, 2016-12-02:
@makruiten Have you tried adding a software mixer to the alsa asound.conf config file?

Fornoth, 2016-12-12:
You can see #11 for software volume mixing