Fornoth/spotify-connect-web #17: Run along side RuneAudio

JozoVilcek, 2015-11-04:
I am trying to get this working on raspberry pi where I have RuneAudio already installed and running. Following `Quickstart using a pre-built chroot` I am able to get it up and running, meaning, I can see device in android Spotify app, send music to it and see actions logged in console. 

However, I can not hear any sound. I can play audio from RuneAudio but can not hear Spotify. Can I get those guys to play nicely with each other and take turns on audio ouptut?

JozoVilcek, 2015-11-04:
I have no idea how stuff around audio output works. I have just figured out, that spotify-connect-web is playing via phone jack on raspberry rather than via my hifiberry digi.

JozoVilcek, 2015-11-04:
When specifying output via `--device` they block each other and I have to manually release ouptut from runeaudio to make it available for spotify-connect and vice versa. Can this be done automatically? To steal output on playback start, which will stop previous client?

maumi, 2015-11-04:
for now, you could only kill one when the other starts and vice versa.
but it's on todo list of fornoth to release audio device when not in use

JozoVilcek, 2015-11-09:
I managed to get this working in #18

Fornoth, 2016-01-15:
Since #18 was merged, is this fixed now?

JozoVilcek, 2016-01-15:
Yes, works for me. Thanks!

closed 2016-01-15 by JozoVilcek