Fornoth/spotify-connect-web #20: No Event if playback queue ends

maumi, 2015-11-19:
If you play only a single song or reach the end of your queue, no event will be triggered.
That's why (sometimes) the audio device isn't released with the pull request from @JozoVilcek and I use the events to power off some things.
I think it hangs because it's waiting for new data, but there is no. In log you will get a "buffer underrun" warning for this.

Fornoth, 2016-01-08:
I can't seem to duplicate this, but there's an event named `SpPlaybackNotifyAudioDeliveryDone` in the newer versions. I'll add support for that after I add support for the newer libraries

Fornoth, 2016-01-15:
I pushed support for the newer libraries (copies are on issue #25), try those and see if it releases the device

maumi, 2016-01-15:
hey, tried with newer libraries and it works. Thanks man.

closed 2016-01-15 by maumi