Fornoth/spotify-connect-web #21: Odroid C1+

tokazio, 2015-11-25:
sudo LD_LIBRARY_PATH=$PWD python main.py -d
Traceback (most recent call last):
  File "main.py", line 9, in <module>
    from connect import Connect
  File "/spotify-connect-web/connect.py", line 8, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/spotify-connect-web/console_callbacks.py", line 16, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
IndexError: list index out of range

maumi, 2015-11-25:
what do you get when typing "amixer" in bash?

tokazio, 2015-11-25:
nothing

maumi, 2015-11-25:
that explains the error. 2 choices: use a softvol mixer, see issue #11 
2. choice: comment any line out using mixer in code in connect.py and console_callbacks.py, but then you can't change the volume.

tokazio, 2015-11-25:
Ok it's working.
I have the same problem that i have with 'scplayer' the sound is stuttering on my TEAC UD501 DAC (OTG). [here](https://github.com/crsmoro/scplayer/issues/15)

Do you think that it's a spotify embbed lib problem?

Because it's working fine with [this DAC](CM108AH DAC)

maumi, 2015-11-26:
My output looks similar. Haven't had any stuttering ever. But I have no Odroid to test. Have used it on a pi 2 (internal chinch, internal HDMI, external cheap usb sound) with OSMC and also on a Cubox i2 with debian. For both devices I use the armhf lib. 
I have no real idea what kind of problem it is. But because it's also a problem in scplayer I think it should be a lib problem or sound card compatibility problem. 
Do you use the DAC otherwise with alsa?

tokazio, 2015-11-26:
Ok i've found! it's an os problem or os configuration problem.
Tried with another image and sounds good with scplayer and openal.
Great work guys! thanks.

closed 2015-11-26 by tokazio