Fornoth/spotify-connect-web #23: chroot: failed to run command ‘bash’: Exec format error

klokop, 2015-12-09:
root@volumio:~/spotify-connect-web-chroot# ./spotify-connect-web.sh --username a --password b
chroot: failed to run command `bash': No such file or directory

Fornoth, 2015-12-09:
What OS/device are you using?

klokop, 2015-12-10:
https://volumio.org on a Raspberry Pi.

Fornoth, 2015-12-10:
Do you know if the OS is softfloat or hardfloat? and RPi 1 or RPi 2? Also, which libspotify_embedded are you using?

djmaze, 2016-01-02:
It's probably the Rocki one because he uses the pre-built chroot script.

isabido, 2016-01-13:
I have the same error. My hardware is RPI2 with Archlinux.

To make it work I have to run the script line by line. Entering the chroot environment. and manually run the python.

In my chroot there is no bash.

djmaze, 2016-01-13:
Okay, I changed the script to use the full path (`/bin/bash`). Should work now.

You need to change the script accordingly or just redownload it:

    curl -O http://spotify-connect-web.s3-website.eu-central-1.amazonaws.com/spotify-connect-web.sh

isabido, 2016-01-13:
Thanks!!!! Now its Ok 😉

klokop, 2016-01-14:
Nope, still getting the same error. On an Ubuntu/Xeon system this time.

djmaze, 2016-01-14:
@klokop spotify-connect-web uses a spotify library from ARM-based receivers and thus works only on systems running the armel/armhf architecture, e.g. Raspberry Pi. (We should probably document that fact better.)

You could try [librespot](https://github.com/plietar/librespot) on your x86 machine instead.

klokop, 2016-01-14:
I figured as much. Will try on Pi. thx.

Fornoth, 2016-03-29:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-03-29 by Fornoth

klokop, 2016-03-29:
Nope, is fine to close. It's all working now. Thanks!

1337sup3rh4x0r, 2016-06-23:
I just ran into the same issue when trying to install this on a debian x64 system.
So can this be resolved by just exchanging a binary within the chroot at some place or does it need more changes than that?

djmaze, 2016-06-23:
@1337sup3rh4x0r This software depends on a spotify library binary which can only be found on ARM-based receiver devices. Thus it is not possible to run this on a x86/x64 system. 

You should probably try out [librespot](https://github.com/plietar/librespot) instead.

1337sup3rh4x0r, 2016-06-24:
Thank you for clarifying. So does the binary within spotify-connect-web need to be replaced with librespot or does librespot offer Spotify connect features and replaces spotify-connect-web as a whole?

djmaze, 2016-06-24:
It replaces it as a whole. There already is a [sample program](https://github.com/plietar/librespot#usage) which acts as a spotify connect receiver.