Fornoth/spotify-connect-web #24: Adio quality with default Raspberry Pi 2 soundcard

wilima, 2015-12-29:
Hey,

is it possible to get better quality with rasp2 sound card? I triend my h2n usb microphone with output (it has soundcard) and the quality was even worse, where can be a problem?

https://www.dropbox.com/s/xb40ikg260f293y/Rasp.m4a?dl=0

Thanks

maumi, 2015-12-31:
Sound quality is and never will be the best on pi because it's PWM sound. 
But if you don't listen to music at verly low volume, the quality could be acceptable. 
At low volumes or at pause you will hear crackling sounds which are really annoying.
But with a 8€ usb sound this sounds go away. Sure it's also not good sound but it's only for my bedroom where I don't listen often to music.
You could try do decrease the volume of the soundcard with alsamixer. Could help for sound quality.

Fornoth, 2016-03-29:
You can try using the `--bitrate` option, but I'm not sure if it actually works

Fornoth, 2016-07-03:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-07-03 by Fornoth