Fornoth/spotify-connect-web #25: raspberry pi 2 web server

vit3k, 2016-01-06:
Hi
On Raspberry Pi 2 I can't access any web page because it gives Segmentation Fault.

Regards
Vit3c

maumi, 2016-01-06:
try to comment out line 87 in connect.py: "print_zeroconf_vars()"

vit3k, 2016-01-06:
Thanks that worked but actually I wanted to run it to check SpZeroConfGetVars because I'm trying to use this in SCPlayer but it always gives me segmentation fault there either. Is library broken?

maumi, 2016-01-06:
I think it's an rights issue. You could also try to run it as sudo with -d (debug mode).
Or have a look at uncommited pull request which tries to fix zeroconf problems.
Can't help with the other problem you have.

Fornoth, 2016-01-06:
The `spotify.h` file needs to be updated for the newer libraries, I'll try to that get done soon

Fornoth, 2016-01-06:
But yeah, commenting out print_zeroconf_vars() should fix the segfault at least

vit3k, 2016-01-06:
Yes it fixes it and web pages can be accessed. Also zeroconf is working until you open spotify on a device with different account logged in. Then it gets segfault too.

plietar, 2016-01-07:
I have a (partially) updated spotify.h here:
https://gist.github.com/b25fddc43e0dc9db6d5a

The main difference is that some SpZeroConfVars's fields are one byte longer (0x41 instead of 0x40)

I use it with the .so from rocki's firmware, from about three weeks ago. The reported version is `release-esdk-1.18.0-v1.18.0-g121b4b2b`

vit3k, 2016-01-08:
I use https://github.com/sashahilton00/spotify-connect-resources/blob/master/armhf%20version/libspotify_embedded_shared.so

and it doesn't look like 0x41 fix is working with it. Where can i get the newest one?

Fornoth, 2016-01-09:
Try this one
[libspotify_embedded_shared.so.zip](https://github.com/Fornoth/spotify-connect-web/files/83472/libspotify_embedded_shared.so.zip)
and to update the header file run these commands (I'll come up with a more permanent solution later)
```
rm spotify.h spotify.processed.h
wget https://gist.github.com/anonymous/b25fddc43e0dc9db6d5a/raw/92275e0354a31bf32ae837d46d80db8d3c4bb727/spotify.h
cpp spotify.h > spotify.processed.h && sed -i 's/__extension__//g' spotify.processed.h spotify.h
```

vit3k, 2016-01-09:
I can't use this lib on rpi2 with OSMC installed.
cffi.ffiplatform.VerificationError: importing '/home/osmc/spotify-connect-web/__pycache__/_cffi__x2b35a451xd84c6704.arm-linux-gnueabihf.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory

I tried to put in local dir and in /usr/lib. It works with hard float one and not with this.

Fornoth, 2016-01-11:
Try this one
[libspotify-embedded-shared.zip](https://github.com/Fornoth/spotify-connect-web/files/85265/libspotify-embedded-shared.zip)

Fornoth, 2016-01-15:
I just pushed a new version, try it with the library in the comment above this one

vit3k, 2016-01-16:
now it's working great. Thanks.

closed 2016-01-16 by vit3k