Fornoth/spotify-connect-web #28: audiodevice blocked while running

Sindaugh, 2016-01-15:
Hello,

first, I love it!

second, not so much as it completely blocks my sound card. I am using a hifiberry on my RPi B - with raspbian and kodi. While spotify-connect is running kodi will not play any sound even though playback is stopped.

Would it be possible to free the device on stopped playback?

Thanks

maumi, 2016-01-15:
should work with one of the newest commits

psych0d0g, 2016-01-15:
else it might be your alsa configuration not using dmix for sharing the sound device across multiple audio sources. have a look here: http://alsa.opensrc.org/DmixPlugin

Sindaugh, 2016-01-15:
Yes, it is most likely the dmix missing. I am just getting it setup.

Me bad. 

Thanks a lot

Sindaugh, 2016-01-16:
Hi there,

just a little feedback in case somebody has the same issue.

This is definetly alsa setup related. Funnily enough kodi is doing the malpractice here. Apparently it can only address pcm.!default from asound.conf if it is type hw. Plug does not work. Meh. I give it a try with a .asoundrc settings.

Another funny thing about Kodi: If you select it without any plugs and then change the settings it works. But do not enter the Audio settings again. It immediately switches to HDMI and my Hifiberry does not show up again.

consider it closed. Thanks for the quick and good help.

Cheers

Sindaugh, 2016-01-21:
Hello,

I have been trying to get this to run via my (now happily device sharing ALSA config with dmix). I setup Kodi correctly and can make speaker-test or run play for a sinus wave successfully. What I can't get to run is spotify. Here is my call, maybe you can advise on it:
nohup ./spotify-connect-web.sh --username johndoe --password foobar--bitrate 320 --name RPi --device default &

I am still in testing mode here. As soon as I have it running I will move it to chron with a little check or maybe systemd. Here's also my asound.conf (which is the same as my .asoundrc). 

#asound.conf
> pcm.hifi_berry  {
 type hw card 0
}
pcm.!default {
        type plug
        slave.pcm "dmixer"
}
pcm.dmixer  {
        type dmix
        ipc_key 1024
        slave {
                pcm "hifi_berry"
                channels 2
        }
}
ctl.dmixer {
        type hw
        card 0
}

Thanks

Sindaugh, 2016-01-22:
Just had a thought. Could it be that spotify is actually locking my default device.

I will create a seperate plug that will forward to dmix for merging.

pcm.python {
type plug
slave.pcm "dmixer"
}

then start spotify with --device python

I'll report back.

maumi, 2016-01-22:
I don't think that this will help. Whole audio device is blocked. 
What is exactly your problem?

Sindaugh, 2016-01-22:
Hello,

my problem is that the hifiberry is blocked by spotify as soon as it starts. No other sources are possible.

With Kodi, speaker-test, play, ... it is possible to multisource on the device. for testing purposes I had a movie running on Kodi and I started a sinus with play. Was weird, but it worked.

What I want to achive:
Have Kodi and spotify share the device. Meaning I can leave Spotify and Kodi running in parallel without having to bother about exclusive locking. Kodi will run into problems if spotify locks the device (crash, disable all sound output).

maumi, 2016-01-22:
So it's spotify blocking and not kodi? For a better co-existence of both you should check in kodi settings, that no system sounds are played and kodi shouldn't keep the audio device active for x minutes.
Do you have the newest commits of spotify-connect?
Especially pull request #18?
With this, spotify should release the audio device when not playing.

Sindaugh, 2016-01-22:
I will check regarding the latest build and adjust Kodi settings.

However that would mean that I need to solve the issue organisationally, right.

Okay. Still I wonder why spotify is unable to share the device as all other applications can...

maumi, 2016-01-22:
yeah, spotify connect can not share audio device ootb. Should be possible anyhow. But I don't see the importance, some (like me) are happy with blocking. When I listen to music, I only want to listen to music without other sounds ;)

Sindaugh, 2016-01-22:
Thanks for the additional idea. I'll go exploring :-)

Sindaugh, 2016-01-23:
Okay.

Harcdcoding the name into console_callbacks will not help. --device is put into the alsa.pcm.card (go figure). A newer version of alsaaudio would support the device option, but this is where my skills leave me. How would I update alsaaudio for spotify?

Thanks

maumi, 2016-01-23:
You can look in my fork. You need pyalsaaudio>=0.8 
Either install via pip or call my requirements.txt
In console-callbacks you need the change from card to device. 
I'm not at home so you need to look for your own.

Sindaugh, 2016-01-23:
Okay. I just updated console_callbacks.py to the latest version. I'll live with it (meaning I was still being told that device was not allowed on pcm...

For me it is closed (again)

closed 2016-01-23 by Sindaugh