Fornoth/spotify-connect-web #30: Monitor status

radem205, 2016-01-18:
Is it possible to monitor the status of Spotify connect, by using for example Monit? And if the service is down, restart the service.

maumi, 2016-01-18:
you can use daemontools for restart.
you can also log with it.
https://cr.yp.to/daemontools.html

radem205, 2016-01-19:
Could you help me with the right command to check the status and restart the service if necessary? Tnx!

herrernst, 2016-01-19:
Or if your system uses systemd, just write a small unit file, systemd will restart a service automatically if it fails.

maumi, 2016-01-22:
with daemontools, you simply need to install it.
http://blog.rtwilson.com/how-to-set-up-a-simple-service-to-run-in-the-background-on-a-linux-machine-using-daemontools/
in run file you need a starting script for spotify. If it fails, it will restart also automatically. 
If it fails on start, it surely won't help.
If you want logging, you can google for that.

cadavre, 2016-05-25:
I used supervisord.org to keep this running.