Fornoth/spotify-connect-web #32: Remote control the this from program

wendelit, 2016-01-27:
Hallo, 
I would like to be able to start and play Spotify songs on this with a program. Is that possibale?

Sindaugh, 2016-02-06:
Well,

as you need a spotify account you can control it via your "other" Spotify device... at least that's what I do.

maumi, 2016-02-06:
Or you can start with http. Yourip:4000 and then play. You also can use http requests to call it with any device

Cesc1986, 2016-02-08:
Hi maumi, I have a similar question. Can you please elaborate on how to use http requests? 

Spotify Connect works so far for and yes, I can start and control it with my iPad, Mac or whatever device in the same LAN. However the idea is to start it headless on my Raspberry, with me controlling just the start of a playlist via an button/script on my apache server. Only when I decide to change the music, I will start my spotify app on my iPad and then control the music on my raspberry via spotify connect.

maumi, 2016-02-08:
afaik you can only use the basic calls via http request. so play/pause, prev/next.
you can use curl or wget on pi with the adress 
YOURIP:4000/api/playback/play 
other commands are pause, previous, next.
If you can use vol up and down I'm not sure at the moment.
If you call play your playlist will start where it last stops.

Cesc1986, 2016-02-08:
I understand ... however that's not exactly what I am looking for. I want to give the command to start a certain song or playlist. Do you think implement such a functionality is possible without too much hassle?

Fornoth, 2016-02-08:
Until librespot is ready, this program can only control itself, and you can't tell it to play a song/playlist/etc without using an external program (something like the spotify client)

You would have to use something like this https://github.com/badfortrains/spotcontrol

Fornoth, 2016-03-29:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-03-29 by Fornoth

axelsimon, 2016-08-23:
Just a quick question, how hard would it be to write a Kodi add-on that would use the API to display what's now playing and offer the basics of playback control? (play, pause, previous and nex, possibly toggling repeat and shuffle)
With my limited knowledge, it looks like the right calls to the API from the Kodi add-on would do just that, without the need for something spotcontrol or librespot, no?

Fornoth, 2016-10-02:
@axelsimon yeah, it should be possible, @noemu posted a Kodi plugin that they started in #70 https://github.com/noemu/script.example-master