Fornoth/spotify-connect-web #33: cffi.ffiplatform.VerificationError

klokop, 2016-01-27:
Any idea as to a solution to below issue?

root@raspberrypi:~/spotify-connect-web# LD_LIBRARY_PATH=$PWD python main.py -d
Loading Spotify library...
Traceback (most recent call last):
  File "main.py", line 10, in <module>
    from connect_ffi import ffi, lib
  File "/root/spotify-connect-web/connect_ffi.py", line 24, in <module>
    libraries=[str('spotify_embedded_shared')])
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 433, in verify
    lib = self.verifier.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 111, in load_library
    return self._load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 222, in _load_library
    return self._vengine.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 155, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/root/spotify-connect-web/__pycache__/_cffi__x4c785defx96dcd04b.arm-linux-gnueabihf.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory

root@raspberrypi:~/spotify-connect-web# ls -las *spotify*
180 -rw-r--r-- 1 root root 183260 Jan 27 09:23 libspotify_embedded_shared.so
  4 -rw-r----- 1 root root    321 Jan 27 09:25 spotify_appkey.key
  8 -rw-r--r-- 1 root root   5529 Jan 27 09:05 spotify.h
  8 -rw-r--r-- 1 root root   8161 Jan 27 09:05 spotify.processed.h

maumi, 2016-01-27:
why not try without LD_L... and copy libspotify_embedded_shared.so to /usr/lib

klokop, 2016-01-27:
Although I hardly know what I'm doing, that' exactly what I did try. Same error though.

(fyi: I got the .so from the chroot dir, I tested previously)

root@raspberrypi:~/spotify-connect-web# ls -las /usr/lib/libspotify_embedded_shared.so 
180 -rw-r--r-- 1 root root 183260 Jan 27 11:31 /usr/lib/libspotify_embedded_shared.so

root@raspberrypi:~/spotify-connect-web# python main.py 
Loading Spotify library...
Traceback (most recent call last):
  File "main.py", line 10, in <module>
    from connect_ffi import ffi, lib
  File "/root/spotify-connect-web/connect_ffi.py", line 24, in <module>
    libraries=[str('spotify_embedded_shared')])
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 433, in verify
    lib = self.verifier.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 111, in load_library
    return self._load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 222, in _load_library
    return self._vengine.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 155, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/root/spotify-connect-web/__pycache__/_cffi__x4c785defx96dcd04b.arm-linux-gnueabihf.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory

maumi, 2016-01-27:
which Pi?
1 or 2?

klokop, 2016-01-27:
Raspberry Pi RBCA000 Mainboard (ARM 1176JZF-S, 512MB RAM, HDMI, 2x USB 2.0, 3,5 Watt)

maumi, 2016-01-27:
afaik you need to use docker or chroot version.
because pi 1 is armsf (soft float).
Or buy yourself a pi 2 ;)

klokop, 2016-01-27:
Ah. Okay, thx.

Fornoth, 2016-07-03:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-07-03 by Fornoth