Fornoth/spotify-connect-web #35: More than one instance of spotify-connect-web

jonofe, 2016-02-21:
I installed spotify-web-connect on two differen RPi2, but only one if them shows up in the devices list of the spotify app. Did I something wrong? Or is that by design?
I'm using the same app-key and the same premium account.
Any hint is very much appreciated.

In addition I experience the problem with the strange volume scaling, i.e. using the HW buttons of my phone/tablet controls the volume not smoothly, but starts at very low volume and increases towards the end of the scale exponentially. Is there any solution for that?

BR
André

maumi, 2016-02-21:
No it should be possible to have as many clients as you want. Ihave 2 running all the time. But for sure you can only listen to one client at a time. 
How do you start spotify connect? Chroot, docker or directly python?
There is one PR for better volume scaling. You can try this. Or you could look in my fork and in console_callbacks.py at the bottom. I used some crude code hack to have a better range.

jonofe, 2016-02-21:
Now I suppose I know why this happened. I just realized that both RPi2 have the same device-id. I think this is caused due to the fact, that I cloned the SD card. 

In which step is this device-id generated? Do I have to install both devices separately?

Will give the new new console_callbacks.py a try. Hope that will improve. Thanks for this advice.

Fornoth, 2016-03-29:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-03-29 by Fornoth