Fornoth/spotify-connect-web #37: Service randomly disappears from network

radem205, 2016-02-24:
Hi,

In some cases the spotify connect service disappears from my network, so I can't find it on other devices in Spotify. The spotify connect web service is still running. I've to manually restart the service to make the service available again.

Is there any possibility to monitor the connection, instead of the service?

maumi, 2016-02-24:
How do you start it? Python? How you start the service? I don't have this issue so I would try to view logs and find the failure to fix it

radem205, 2016-02-24:
On the Rpi 2 I use wifi to connect and the following command:

./spotify-connect-web.sh --name LivingRoom --username 12345678 --password xyz123 --bitrate 320

I also use the wifi checker on the site below to monitor the wifi connection: https://github.com/dweeber/WiFi_Check/blob/master/WiFi_Check

What is the best way to start and monitor the service?

Thanks for your fast reply!

maumi, 2016-02-24:
First I would try to start it without chroot.
You need to copy the spotify lib from here
https://github.com/sashahilton00/spotify-connect-resources/tree/master/armhf%20version
to /usr/lib
and you need to install things from requirements like said in readme.
Then you should be able to start with
python main.py --your parameters

Another small security issue. If you write your account as parameter it could be read very easy. Would be better to not type it there and then log in at pi-IP:4000

radem205, 2016-02-24:
Thanks I'll try that!

Another side note: I use it together with Volumio 1.55. Is it better to run it stand alone on a rpi?

maumi, 2016-02-24:
As it's working (mostly) there shouldn't be a problem with volumio. But I haven't ever used volumio, so can only guess. Only thing that could be issues with access of the sound card.

radem205, 2016-02-24:
Ok, thanks for your investigation. I'm not familiar with Linux, so which clean debian-based image can I use to create a clean install of the spotify connect? I'll only use Spotify so for me it's maybe better to have a clean install.

maumi, 2016-02-24:
Yep Raspbian jessie would be then my choice. And if you also need other things like airplay you could easily install other software on raspbian like shairport-sync.

radem205, 2016-02-24:
Thank you so much for your help and time! I'll let you know if a fresh install is the solution.

One question: to use the service without chroot (like you said), I've to install the whole package as explained in the readme under "Quickstart using a pre-built chroot" and combine that with the instructions in your earlier post, right?

maumi, 2016-02-24:
You only need the Installation part.
You could also look in the file Dockerfile, there you can see the necessary commands.
But for sure, always without the first word.

radem205, 2016-02-24:
Sorry for my stupid questions, but do I have to copy the whole Sporify Connect Web repro to the folder 'spotify-connect-web' in the root and execute the commands from there?

maumi, 2016-02-24:
yep. you don't need the whole files but it's the easiest to download repo, unzip and put folder onto pi per sftp.
Or easier one, if you first install git by 
sudo apt-get install git-core
then you could do
git clone https://github.com/Fornoth/spotify-connect-web

radem205, 2016-02-27:
I've tried your instructions, with succes. But, when I close the Putty console the connection disappeared from the network. Also ip-address:4000 is down.

What can cause this problem?

And the output of aplay -l looks like this:
---------------------------------------------------------------
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
---------------------------------------------------------------

Do I have to set the device param to 1 to work with the hifiberry digi+ card?

Tnx again for your reply.

maumi, 2016-02-27:
Easy fix is to add a & after the python command to put the program in background. Maybe you also need a nohup before command. 
Better would be to install and use daemontools. You can google it. Can be installed by apt-get.
To use hifiberry I think you need to install a version of pyalsaaudio minimum 0.8 but then you need to change a line in console_callbacks.py. 
Sounds more complicated then it is. You can also look in my fork of spotify connect and in the other issues, also the closed ones. 
If you are stuck, even after many reading and trying you can ask again for help.

radem205, 2016-03-01:
Thanks to your help I have something up and running. But unfortunately I still have no sound. 
Spotify connect is now monitored by using Daemontools. This working great. 

The problem is that my hifiberry digi+ isn't recognized by Spotify Connect. I'm getting the following error when I try to play a song:

ALSA lib conf.c:4705:(snd_config_expand) Unknown parameters CARD=sndrpihifiberry
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=sndrpihifiberry

And I'm using the following command to run Spotify Connect (for testing):
sudo python main.py  --device sndrpihifiberry --name Woonkamer --username radem205 --password ayhadr9A --bitrate 320 --key /home/pi/spotify-connect-web/spotify_appkey.key

The output of aplay -l is:
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

When I run the command "$peaker-test -Dplughw:sndrpihifiberry -c2" I get noise out of the speakers, so it seems that the card is working properly.

Which name do I have to use for the --device param? Or do I have to change something in de console_callbacks.py (I tried changing card = self._args.device) to device= self._args.device), without success)?

Thanks in advance!

maumi, 2016-03-01:
ok, I do the cooking for you. 
In putty/console:
pip install pyalsaaudio>=0.8
if it fails, try again with sudo.
Then you can try to start spotify-connect, should fail with failure that pcm.card is not listed or so.
So you need to change line in console_callbacks.py from card = self... to
device = self....
Then it should work. But I couldn't managed myself to use 2. audio device by parameters. So I hard coded it in cons*_callb*.py.
I'm not certain I think output was working but not the volume adjust.
I've added a softvol for my second sound device.
For that you need to create a file by
sudo nano /etc/asound.conf
In nano, you need to copy sth. like this (maybe you need to change device)

pcm.softvol {
    type            softvol
    slave {
        pcm         "default"
    }   
    control {
        name        "Master"
        card        1   
    device      0   
    }   
}

pcm.!default {
type hw
card 1
device 0

after that, save by ctrl+x.
You could try if its working with speaker-test -Dsoftvol and also without -Dsoftvol you should get noise.
If that is working you can override the mixer call in cons*_callb*.py
Near to line 90 you need to change mixer = alsa.mixer* to
mixer = alsa.Mixer(control="Master",cardindex=1)
but that's only necessary if you need volume adjusting.
Otherwise you can call spotify-connect with --device softvol or hardcode softvol in the newly changed device line.

radem205, 2016-03-01:
I did al of that already (but thanks to explain). Unfortunately the Hifiberry Digi+ doesn't support software volume control (as far as I know). Your instructions only works if the card supports software volume control, right? After my previous post I did some test and this is what works for me:

console_callbacks.py:
card = self._args.device) to device= self._args.device)

And set -Dplughw:sndrpihifiberry as param. This solved my problem.

maumi, 2016-03-01:
If it has no softvol you surely don't need softvol.
But you could change the volume in volumio with software (app or whatever) for you digi+,right?
If yes, try asound.conf with only 
pcm.!default  {
 type hw card 0
}
ctl.!default {
 type hw card 0
}
It overrides defaults. Then you shouldn't need -D param.
And hopefully volume adjustment works.
You could also test and play around with
alsamixer
or if it's not starting
alsamixer -c1
So you get now sound out of your digi+?

radem205, 2016-03-01:
I don't use Volumio anymore. I installed Spotify Connect from Raspbian Lite. 

I get sound out of my digi+ now. Thanks. Unfortunately, softvol doesn't work. So I've to change the volume by using the remote controller of the DAC. 

Thank you so much for your support. I appreciate that!

maumi, 2016-03-01:
Yep, I know. My question was: Was it possible with volumio to change volume or where you forced to use your remote of your DAC?
If your problem is solved, can you close the issue?

radem205, 2016-03-01:
I didn't connect the rpi to my DAC, because I was just testing if I can get Volumio and Spotify up and running on the rpi. Yesterday,, I connected the rpi to my DAC to see (and hear) if everything works :). So I can't answer your question.

I'll close the issue.

radem205, 2016-03-01:
I reopened this issue, because I have something strange:
When Spotify Connect Web is running I can't access the rpi with the corresponding ip address. Even pinging the rpi doesn't work. But, the Spotify Connect still works. I can play music. 

Do you know what can cause the connection to the rpi is lossed, while Spotify Connect still works?

maumi, 2016-03-01:
explain "runnning". When you play music or even when it's started and running.
If so, how do you stop the daemon without access?

radem205, 2016-03-01:
Running in terms of play, skip and pause music. So I have fully controll of the music using the Spotify app. It looks like the ip address is lossed after a couple of minutes. Now I can't stop the daemon. A restart of the rpi doesn't solve the problem; I can access the rpi, but after a while the ip address isn't visible in the network.

maumi, 2016-03-01:
strange. Post your daemontools run-file.
Maybe try it for testing without daemontools only with &.

radem205, 2016-03-01:
run file:

--------------------------------
#!/bin/sh
exec 2>&1
exec /usr/bin/python /home/pi/spotify-connect-web/main.py -Dplughw:sndrpihifiberry --name Woonkamer --username ***--password ***--bitrate 320 --key /home/pi/spotify-connect-web/spotify_appkey.key
--------------------------------

/etc/network/interfaces

--------------------------------
auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp

--------------------------------

maumi, 2016-03-01:
Can't see a problem. Try to comment out 
#exec 2>&1
and before exec 
cd /home/pi/spotify-connect-web
then try to run with --bitrate 320 and -d (debug) only.
exec /usr/bin/python main.py -Dplughw:sndrpihifiberry --name Woonkamer --bitrate 320 -d

How is pi connected to network? wired or over wifi?

radem205, 2016-03-01:
The pi is connected over wifi. Maybe a stupid question; but where can I see the debug output?

maumi, 2016-03-01:
yep, you should find your output there. debug means only that more infos are written to stdout (which is logged by multilog).
I run and use Spoti Connect with only these params, so yes it's working.
But you need to log in once on ipOfPi:4000 so that a credentials file is created.
You need to test more around because the failure could be anything.
You could try to comment out at line 101 in connect.py
print_zeroconf_vars()
seems to create problems for some.
What I would do is to nail things down to a minimum possible failures. So connect via LAN. Start spoti connect directly without daemontools, and so on.
If you are sure it's working and pi isn't loosing connection then you can reverse your changes step by step.

radem205, 2016-03-01:
Ok, I'll do that. Thanks for your effort! Once I found the solution, I post it here. In the mean time I close this issue.

closed 2016-03-01 by radem205