Fornoth/spotify-connect-web #38: Connected family account don't see device

superroma, 2016-02-24:
I set up spotify Family plan for my family members.
It seems only one account can see device with spotify-connect-web - the one which generated a key
In contrast, my Marantz receiver is visible for any spotify premium user in the same network

CppNigger, 2016-03-02:
Make sure you are using discovery-mode and not inputting any username/password.

Fornoth, 2016-03-09:
You need to use zeroconf for it to show up for other people other than the logged in account https://github.com/Fornoth/spotify-connect-web#logging-in

balbuze, 2016-05-29:
Could you please give more details on how make it works... I can see device with my "master" account. I use avahi command as explain. But other family devices still doesn't see it... Do it need to change the command line (removing credentials - I tested and it still connect) exept from other account.

q20, 2016-07-04:
I, too, wouldn't mind detailed instructions on how I accomplish being able to see the Spotify Connect device from arbitrary accounts.

Fornoth, 2016-07-04:
What device are you trying to access the connect device from?
Also, try using the new packaged release, since the chroot hasn't been updated in a while https://github.com/Fornoth/spotify-connect-web#quickstart-using-a-packaged-release

Fornoth, 2016-07-04:
Also, try one of these apps after you do avahi-publish-service, and see if you see something about spotify connect
Android: https://play.google.com/store/apps/details?id=com.druk.bonjour.browser&hl=en
iOS: https://itunes.apple.com/us/app/discovery-bonjour-browser/id305441017?mt=8

balbuze, 2016-07-04:
I use a raspberry pi2 with volumio 1.55 and spotify connect . It works perfectly with premiuim account. Since I have a family account, it still works with my account, but other devices can't access to spotify connect (with official app).  See a screenshot took from one of device that not work. with bonjour browser. As you can see spotify connect is running.
[image]

Fornoth, 2016-07-04:
Are you using the chroot version?

balbuze, 2016-07-05:
Yes chroot version

Fornoth, 2016-07-05:
Can you try the packaged version? https://github.com/Fornoth/spotify-connect-web#quickstart-using-a-packaged-release

balbuze, 2016-07-05:
Thank you. I'm going to test. I keep you in touch.

balbuze, 2016-07-05:
Ok thank you x1000 !!! It works perfectly now !
Now the next step is to intergate it in Volumio 2 as a plugin...

balbuze, 2016-07-05:
But I can't connect to IP:4000 via a web browser, I get a Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
or in console
192.168.0.8 - - [2016-07-05 20:18:58] "GET /login/_zeroconf?action=getInfo HTTP/1.1" 200 612 0.025204
[2016-07-05 20:19:47,681] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "flask/app.py", line 1988, in wsgi_app
  File "flask/app.py", line 1641, in full_dispatch_request
  File "flask_cors/extension.py", line 188, in wrapped_function
  File "flask/app.py", line 1544, in handle_user_exception
  File "flask/app.py", line 1639, in full_dispatch_request
  File "flask/app.py", line 1625, in dispatch_request
  File "/home/pi/spotify-connect-web/main.py", line 66, in index
    return render_template('index.html')
  File "flask/templating.py", line 133, in render_template
  File "jinja2/environment.py", line 851, in get_or_select_template
  File "jinja2/environment.py", line 812, in get_template
  File "jinja2/environment.py", line 774, in _load_template
  File "flask/templating.py", line 57, in get_source
  File "flask/templating.py", line 85, in _get_source_fast
TemplateNotFound: index.html
192.168.0.9 - - [2016-07-05 20:19:47] "GET / HTTP/1.1" 500 412 0.084124
[2016-07-05 20:19:54,628] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "flask/app.py", line 1988, in wsgi_app
  File "flask/app.py", line 1641, in full_dispatch_request
  File "flask_cors/extension.py", line 188, in wrapped_function
  File "flask/app.py", line 1544, in handle_user_exception
  File "flask/app.py", line 1639, in full_dispatch_request
  File "flask/app.py", line 1625, in dispatch_request
  File "/home/pi/spotify-connect-web/main.py", line 66, in index
    return render_template('index.html')
  File "flask/templating.py", line 133, in render_template
  File "jinja2/environment.py", line 851, in get_or_select_template
  File "jinja2/environment.py", line 812, in get_template
  File "jinja2/environment.py", line 774, in _load_template
  File "flask/templating.py", line 57, in get_source
  File "flask/templating.py", line 85, in _get_source_fast
TemplateNotFound: index.html

Not very important for me...

q20, 2016-07-06:
Could some please explain exactly how to get this working? I've not yet tried the packaged release (will do soon) but even then, I don quite understand how to enable zeroconf.

What I've done up until now is run:
./spotify-connect-web.sh --username 12345678 --password xyz123 --bitrate 320 --name RasPIiConnect
This gives me Spotify Connect for a single account.

Where do I go from here? :)

Fornoth, 2016-07-06:
@balbuze For now, run `./spotify-connect-web` in the same directory that extracted it. It's because it uses a relative path for `index.html`
@q20 use the command `avahi-publish-service TestConnect _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf` 
https://github.com/Fornoth/spotify-connect-web#logging-in (I'll make that section in the README a little clearer)

Fornoth, 2016-07-07:
what options are you using to launch it?

balbuze, 2016-07-07:
I set just username / password and bitrate. I had no time to test other options...

Fornoth, 2016-07-09:
Can you add a `print pcm_args` right before https://github.com/Fornoth/spotify-connect-web/blob/master/console_callbacks.py#L61 and give me the output?

balbuze, 2016-07-09:
hi!
the line is                 pcm = alsa.PCM(**pcm_args)
But it works now after adding --playback_device hw:2
I start to write a plugin for volumio2 using your work : https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect
far from beeing finished... It 's just provide a easy way to install configure and launch spotify-connect-web. It works, but at this point no album art or working buttons in volumio