Fornoth/spotify-connect-web #40: Zeroconf : device name, account ad device type truncated (wrong spotify.h offsets)

Alex131089, 2016-02-29:
With the commit setting offets to 0x41 for zeroconf in spotify.h, those names are truncated : 
```
remote name: drus
account req: MIUM
device type: NOWN
```

It is affecting the name displayed by Spotify (and I guess it's doesn't like the other things too). Reverting to 0x40 is ok : 
```
    char deviceId[0x40];
    char activeUser[0x40];
    char remoteName[0x40];
```

By the way, you say "spotify.h was taken from from https://github.com/plietar/spotify-connect" but I couldn't find it, can you tell me more ?

Fornoth, 2016-03-09:
Which version of the library are you using? And yes, spotify.h was originally taken from that repo (https://github.com/plietar/librespot/blob/old/common/spotify.h ), but the updated version is kept in this repo

Alex131089, 2016-03-09:
You mean version of "libspotify_embedded_shared.so" ? The one included in the pre-build chroot. The test was on RPi.
Why is the new version using 0x41 size ?
Thank you for the information about spotify.h, I guess the readme needs an update :)

Fornoth, 2016-04-02:
Try replacing `/usr/lib/libspotify_embedded_shared.so` with one of these libraries (Use the armv6 ones for Rpi1, armv7 for Rpi2/3) https://github.com/sashahilton00/spotify-connect-resources/tree/master/libs

Alex131089, 2017-05-07:
Hello, 
I bought a Mi Box (MDZ-16-AB), so I'm not going to continue bother with "hacks" for the moment, and I'm not testing, sorry. I really appreciated the ability to use Spotify Connect thank to your work and the work of the others, so thank you (and disappointment about Spotify).
Nobody else seems to have encountered the issue (?), but the state of the issue is up to you :)

oskarn97, 2017-05-20:
Hey guys!
I have the same issue, my device "Wohnzimmer" is being truncated to "hnzimmer"...
Thanks for your help!