Fornoth/spotify-connect-web #42: Set Access-Control-Allow-Origin Header for Webserver

Venice-89, 2016-03-06:
Hi,

it would be nice if you implement the Access Control Allow Origin Header for the internal Webserver.
I pull the META information from the API via jQuery for my Magic Mirror and had to implement this Workaorund into main.py:
http://flask.pocoo.org/snippets/56/

 Without the workaround I got the following error message when I tired to access the API via jQuery:
> No 'Access-Control-Allow-Origin' header is present on the requested resource. 
> Origin 'http://localhost' is therefore not allowed access

Fornoth, 2016-03-09:
I'll see about adding this soon, although it'll probably be in the form of something like a `--cors <host>` parameter

Fornoth, 2016-03-31:
This should be added in the latest commit now 
https://github.com/Fornoth/spotify-connect-web/commit/16d8a898e4ebd8c464bdb73b68ddeb71ba895234
```
Add the --cors option for CORS suppport for the api
The hostname is specified in the format <protocol>://<hostname>:<port>
Example: http://google.com
```

Fornoth, 2016-07-03:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-07-03 by Fornoth