Fornoth/spotify-connect-web #43: AlbumCover Image Size

bastardop, 2016-03-12:
Is there any way to get a bigger AlbumCover Image Size?
i found in utils.py the line
` lib.SpGetMetadataImageURL(uri, lib.kSpImageSizeSmall, image_url, ffi.sizeof(image_url))`

but even if i change it to  `lib.kSpImageSizeLarge` there is no change in the size of the resulting Album Cover File. What did i wrong?

Fornoth, 2016-03-29:
The file probably is increasing in size, it's just the page that limits the display size (https://github.com/Fornoth/spotify-connect-web/blob/master/templates/index.html#L52)
I'll add an option for a larger player

mrdago, 2017-12-29:
I'm also looking to get a bigger AlbumCover Image Size. Unfortunately I'm not a programmer  to implement the promised change.