Fornoth/spotify-connect-web #44: No Sound from USB Soundcard

lukasbenOSL, 2016-03-13:
hey, 

I have a shairpoint script running as well. Output is working with that. with spotify connect running the output is no longer working. 

Any Ideas? 

Setup: 

1. Pi Zero
2. Rasbian Lite
3. External USB Sound Card C-Media USB Headphone Set

alpay -l output: 

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Set [C-Media USB Headphone Set], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

Maaatiii, 2016-03-29:
I've got not identical but similar issue with my USB DAC, it partially works but there was terrible noise sound. I provide small change when i run spotifyconnect i use parameter --device plughw:1 instead of --device hw:0 and it works like a charm. Try it maybe it helps.

lukasbenOSL, 2016-04-06:
thanks!

it is working that way.

closed 2016-04-06 by lukasbenOSL