Fornoth/spotify-connect-web #46: Question: launching script when playback start is reveived and vice versa

quenthal, 2016-03-17:
Any idea if at least first one of the following is feasible:

1: launch script from apotofy connect web when playback starts. when play command is received from some spotify client to device running spotify connect web, one could launch script of their choosing. I'd like to automatically either turn on (with right input) my receiver and speakers or my correct Logitech Harmony activity. This could be done for example via script.

2: using script / http post,put / to send stop playback. I'd like to run this command, for example when my receiver or harmony is asked to change input or activity to non-spotify one.

maumi, 2016-03-18:
2. already possible via http post.

1. possible in my fork of spotify-connect.

quenthal, 2016-03-18:
1: thanks! Checked put your fork, quite self-explenetary!

2: already possible.. Could you give pointers to this?

maumi, 2016-03-18:
sure. PI-IP:4000/api/playback/prev
possible commands: prev, next, pause, play

tatoosh, 2016-04-06:
I also want to use a script, so i can stop/start mpd playback.

flori4n, 2016-07-12:
Hello,
If you want a not perfect hack but working way to know if there is sound on the speaker (rpi) you can try this:

`if grep -q RUNNING /proc/asound/card0/pcm0p/sub0/status 2>&1; then`
      `echo "Playing"`
`	else`
`	echo "NotPlaying"`
`	fi`

I'm using it to start my amplifier when I start any sound, it's working well

tatoosh, 2016-07-12:
Cool hack, thx for sharing

Fornoth, 2016-07-17:
For a spotify-connect-web specific solution, you should be able to get this stuff from the API (/api/info/metadata and /api/info/status)
The full URL would be something like http://localhost:4000/api/info/metadata

seniorgod, 2016-10-15:
It would be great to implement the requested start and stop script calling functionality. In this case i e able to start playing a song on spotify and "release" the ALSA hardware device, which is used by mpd (music player daemon).

flori4n, 2016-10-21:
It's implemented, You can get a JSON object:
http://[yourIP]:4000/api/info/status

seniorgod, 2016-10-21:
Hi,

yes i found that, but that's not what solve my problem....
Until now i used to run 2 audio services on my rasberry pi via my Project DA-converter/preamp: mpd and shairport-sync
Now, with change to spotify, i have three services :-) 
What i "build" for the first two of them was an "easy" way changing the services whitout to have to "click" so much commands on my phone ... 
What does it means? If i hear some hi-res material or some FLAC-files via mpd, the mpd grabs the ALSA hardware-device. I can't change to shairplay (apple airplay) without stopping mpd playback, because mpd block the alsa-device. In shairplay-sync, you can run scripts before start of playing and after end of playing. The script i wrote for "before start of playing" stops playing music from mpd. These stop cause the release of alsa-hardware-device. Because of this, i'm able top simply start playing via airplay whithout the need to stop the mpd-playing before. It will be done by the script. At stopping of airplay, there is also an automatically running script which start playing of my favority radio station via mpd. So i need fewer start and stop oprations on my phone. It would be nice to have the same possibility in spotify-connect-server.  I have no idea how to stop mpd (and release the alsa device) by polling the status, of spotify connect. I guess, spotify cant start playing, if the alsa-device is busy and can't open by spotify-connect. 

best regards

beardedgeek, 2017-07-28:
Did you get anywhere with this? I want to do something similar, but I want the device running spotify-connect to run a script and turn a small local amp on through a relay.