Fornoth/spotify-connect-web #47: Zeroconf responding 500 Internal Server Error on RasPi1

bastardop, 2016-03-17:
Hello

i set up a Spotify-Connect-Web on my RaspberryPi 1. When i tried to set it up via Zeroconf i saw nothing in my Spotify Client. When starting Spotify-Connect-Web with -d Flag i get a 500 Internel server Error when asking for the Route login/_zeorconf?action=getInfo the Traceback is showing up to this error

`    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "main.py", line 39, in wrapped_function
    resp = make_response(f(*args, **kwargs))
  File "main.py", line 223, in login_zeroconf
    return get_info()
  File "main.py", line 243, in get_info
    'brandDisplayName': ffi.string(connect_app.init_vars['brandName']),
AttributeError: Connect instance has no attribute 'init_vars'` 

the i tried to set all ffi. strings ind getInfo() on a static value. So it get past the 500 Error but also trigger an error for login/_zeroconf POST addUser

`    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "main.py", line 39, in wrapped_function
    resp = make_response(f(*args, **kwargs))
  File "main.py", line 229, in login_zeroconf
    return add_user()
  File "main.py", line 269, in add_user
    connect_app.login(username, zeroconf=(blob,clientKey))
NameError: global name 'username' is not defined`

normal Login is working on the Connect-Web and playing and control spotify also works just zeroconf is throwing errors

hope someone can help me

Fornoth, 2016-03-31:
Can you try updating to the latest version and see if that works? Also, which `libspotify_embedded_shared` version are you using?

bastardop, 2016-04-03:
just tried the latest version that ships with the decker installation on my raspi3 with the included libspotify and got the same error

bastardop, 2016-04-19:
so rpi3 in the latest Version without Docker ist working with the armhf libspotify, will try rpi1 in the next days

bastardop, 2016-04-19:
so tried on raspi1 with docker version provided in the readme.md and got the 500 error again on zeroconf

`192.168.5.183 - - [2016-04-19 19:38:31] "GET /login/_zeroconf?action=getInfo HTTP/1.1" 500 412 0.107971`

Fornoth, 2016-07-03:
I made a new type of release since the chroot hasn't been updated in a while (although it doesn't work with Rpi 1/Zero for right now)
https://github.com/Fornoth/spotify-connect-web#quickstart-using-a-packaged-release
I tested Zeroconf, and it seems to be working

Fornoth, 2016-09-04:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-09-04 by Fornoth