Fornoth/spotify-connect-web #48: Redirect Sound to FIFO

bjwelker, 2016-03-29:
Is it possible to redirect the Sound output to a fifo so spotify-connect-web could be used with https://github.com/badaix/snapcast for multiroom audio ?

Fornoth, 2016-03-29:
I'll see about adding a FIFO output option, but in the meantime you can try adding something like this to /etc/asound.conf
```
pcm.!default {
    type file               # File PCM
    slave.pcm "hw:0,0"      # This should match the playback device at /proc/asound/devices
    file /tmp/audio         # Output filename
    format raw              # File format ("raw" or "wav")
    perm 0666               # Output file permission (octal, def. 0600)
}
```
(From http://ninetimes.cat-v.org/tips/2008/07/11/1_Redirecting_audio_output_over_a_network_with_Inferno/)

bjwelker, 2016-03-30:
I came up with the same solution in the night :) Thanks.
A FiFo output would be nice cause snapcast doing a great job on sync audio

breadstuff, 2016-07-28:
Hello everyone.

I just want to mention, that I had a problem if I define `pcm.!default` like stated above, because `default ` seems already beeing taken by the standard configuration in raspbian on a raspberry 2.

My solution for snapcast is `pcm.!snapcast` like in my post [https://github.com/badaix/snapcast/issues/45](https://github.com/badaix/snapcast/issues/45).

Have a nice day with multi-room spotify!

B.

twenta, 2016-09-19:
Hi,

i used the asound.conf posted above. I get really distorted sound in snapcast. I "debugged" as written here: https://github.com/badaix/snapcast/issues/59
When I used aplay and asound sample rate 44100, the sound was like I hear the output from spotify-connect. So I resetted to default 48000, but sound played from spotify is still distorted. Any ideas? I would appreciate some clues, how to proceed.

thanks

coda4k, 2016-09-30:
Does anybody know how to make it work with the chroot version? It seems snapserver is writing to my system's /tmp/snapfifo, but when I start the chroot version, it starts writing to my new chroot /etc/snapfifo, that is not a FIFO, just a regular file.

Thanks in advance!

Fornoth, 2016-10-02:
@coda4k just put the fifo inside the chroot
For example, if your chroot is at ~/spotify-connect-web-chroot/, then put it at ~/spotify-connect-web-chroot/etc/snapfifo on the host, and /etc/snapfifo in the chroot