Fornoth/spotify-connect-web #49: Running as daemon on startup

AlexGustafsson, 2016-04-02:
_First of all. Amazing work!
This is the first solution that works flawlessly or even at all that fully incorporates with Spotify Connect that I am yet to found (hours spent researching and testing the solutions). Volumio works, but I wasn't looking for a complete OS this time._

**How do I run Spotify Connect Web as a daemon automatically on boot?**

AlexGustafsson, 2016-04-03:
### The solution

Create a shell script to hold your start command
`nano /home/pi/spotify-connect-start.sh`
```bash
#!/bin/bash
#Run your Spotify Connect script with your credentials
/home/pi/spotify-connect-web.sh --username USERNAME --password PASSWORD --bitrate 320 --name NAME
```

Make the file bootable
`chmod +x /home/pi/spotify-connect-start.sh`

Add the file to crontab
`crontab -e`

Scroll to the bottom and append
`@reboot /home/pi/spotify-connect-start.sh`

closed 2016-04-03 by AlexGustafsson

Fornoth, 2016-04-13:
@AlexGustafsson After the first login, there's a file named credentials.json created that stores a token sent by spotify to automatically login, so you don't need to have your username and password in the script

ckrp25, 2016-04-20:
@AlexGustafsson Hi- I can't quite get this to work and am wondering what could be going wrong. I followed your steps but I can't get the shell script to run on reboot. It works fine if I just click "spotify-connect-start.sh" and then "execute in terminal". But when I try rebooting it seems that crontab will not run the task. I have confirmed that the crontab file is saving - it still shows up how I left it when I go back into editor, but just won't run the script. Any help would be greatly appreciated!

maumi, 2016-04-20:
Try to add a blank line in crontab after command.

ckrp25, 2016-04-20:
@maumi That didn't help either

maumi, 2016-04-21:
have you done the step where you make your script executable? (chmod +x)
you could try to add 
/home/pi/spotify-connect-start.sh
in /etc/rc.local
at the end before exit 0

ckrp25, 2016-04-22:
@maumi No luck. It's weird, if I look in the syslog I can see where cron runs the command but then nothing happens. Could it be because it tries to do it before I am connected to wifi?

When I run /home/pi/spotify-connect-start.sh in terminal it works fine.. though it does give this error in the syslog:
raspberrypi org.gtk.Private.UDisks2VolumeMonitor[855]: index_parse.c:191: indx_parse(): error opening /home/pi/spotify-connect-web-chroot/dev/BDMV/index.bdmv

Any other thoughts?

mcduarte2000, 2016-07-09:
I have it working. On my script I wrote:

```
cd /home/spotify-connect-web
/home/spotify-connect-web/spotify-connect-web -o hw:0 -n "Quarto Daniel" -b 320 -k /home/spotify-connect-web/spotify_appkey.key
```