Fornoth/spotify-connect-web #50: No volume control on Hifiberry DAC+

whumann, 2016-04-03:
First of all I want to join those saying how fantastic it is to be able to Spotify-Connect on the Raspi. Thanks!

Went back to the 2016-01-19 version because of the problems mentioned [here](https://github.com/Fornoth/spotify-connect-web/commit/b95bda2#commitcomment-16943832), also had to do the `mixer_volume = 65535 # int(mixer.getvolume()[0] * 655.35)` change in connect.py because I'd get a `can't convert negative number to unsigned` otherwise.
So now it runs. Sound comes out of my Hifiberry DAC+, no problems there. Only thing is I can't control volume, neither from Spotify on my PC nor from the webinterface on :4000. I do see messages like `playback_volume: 51773` when I try to adjust volume so the remote-control part works o.k.

aplay -l:
```
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
 /etc/asound.conf:
```
pcm.!default  {
 type hw card 0
}
ctl.!default {
 type hw card 0
}
```
Any idea what I should try next?

maumi, 2016-04-03:
you commented out the mixer volume settings and set it to a fixed value. So you can't change the volume with spotify_connect.
Please add output for 
aplay -L
and for amixer
And try to uncomment the mixer line and start spotify with -m PCM

whumann, 2016-04-03:
Thanks for your suggestions.
I thought line 88 in connect.py is just an initial setting and subsequent volume changes are handled by `playback_volume()` in console_callbacks.py. The reason for commenting out is that apparently `mixer.getvolume()[0]` returns a negative number on my raspi. Any idea what to do with that line because leaving it uncommented as it is doesn't work.

When I start spotify-connect-web.sh (is that what you meant by 'start spotify'?) with -m PCM I get:
alsaaudio.ALSAAudioError: Unable to find mixer control 'PCM',0

aplay -L:
```
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus,
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus,
    Direct sample mixing device
dsnoop:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus,
    Direct sample snooping device
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus,
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus,
    Hardware device with all software conversions
```
amixer:
```
Simple mixer control 'DSP Program',0
  Capabilities: enum
  Items: 'FIR interpolation with de-emphasis' 'Low latency IIR with de-emphasis' 'High attenuation with de-emphasis' 'Fixed process flow' 'Ringing-less low latency FIR'
  Item0: 'FIR interpolation with de-emphasis'
Simple mixer control 'Analogue',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 1
  Mono:
  Front Left: Playback 1 [100%] [0.00dB]
  Front Right: Playback 1 [100%] [0.00dB]
Simple mixer control 'Analogue Playback Boost',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 1
  Front Left: 1 [100%] [0.80dB]
  Front Right: 1 [100%] [0.80dB]
Simple mixer control 'Auto Mute',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Auto Mute Mono',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Auto Mute Time Left',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Auto Mute Time Right',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Clock Missing Period',0
  Capabilities: enum
  Items: '1s' '2s' '3s' '4s' '5s' '6s' '7s' '8s'
  Item0: '1s'
Simple mixer control 'Deemphasis',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Digital',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 207
  Mono:
  Front Left: Playback 196 [95%] [-5.50dB] [on]
  Front Right: Playback 196 [95%] [-5.50dB] [on]
Simple mixer control 'Max Overclock DAC',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock DSP',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock PLL',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 20
  Mono: 0 [0%]
Simple mixer control 'Volume Ramp Down Emergency Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Emergency Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '4dB/step'
Simple mixer control 'Volume Ramp Down Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '2dB/step'
Simple mixer control 'Volume Ramp Up Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Up Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '4dB/step'
```

maumi, 2016-04-03:
ah ok didn't read exactly. The line in connect.py is to get the old volume value. Could be commented out but when you can't get the volume, you also can't set it correctly.
You could try to start spotify (yes spot.con.sh) without -m so that it's working without volume control.
Then find out with 
alsamixer
and testing out the mixers which one adjusts the sound. I think you will see analogue and digital and I would guess depending on how you connected your AVR to your DAC that one of them will change the volume when you are playing a song on spotify.
After that, try again -m flag with the right mixer.
I have no HifiBerry, so it's hard to debug.
Which Pi do you use, what OS (Raspbian Jessie? ) and you use chroot-method to run spotify-connect?

whumann, 2016-04-03:
It's a Raspi2 with Jessie Light (newly setup this weekend, was on wheezy until then)
Ok, without -m and with the commenting out in place I have it running again. 
The DAC+ just has an analogue out with 2 chinch connectors, so there's not much option how to connect it...
In alsamixer I have "Analogue" but that seems to have just to positions: 100 (0 dB gain) and 0 (-6 dB gain), so it's not of much use. I also have "Digital" which gives me good volume control (100 to 0 with 0.5 dB per step at higher volumes, coarser steps at lower volumes).

Hey! just tried it with -m Digital -- and it works. It's so easy once you do it right ;-) Thanks for your help.

The scaling it a bit weird. Midscale on the spotify volume control results in -54 dB which is almost mute, but I think I can fix that myself.

maumi, 2016-04-03:
Yep, that's "normal". You can fix it with the uncommitted PR in this project.
Or dirty fix it like I did in my fork of spotify-connect. Only with if's to use your own scale.

whumann, 2016-04-03:
Well, at first glance that PR looks like a linear remapping -- which I would not expect to undo what I see here, which is somewhat 'warped'. I did a little table mapping what goes into `mixer.setvolume()` to what I ends up in alsamixer. First I thought it looked like a x² and tried a sqrt to correct it. That was a improvement but not perfect. Then I saw, that a step of 10 in setvolume made a constant factor in alsamixer so it's an exonential rule. My current solution is:
`mixer.setvolume(int(100 + 25*log(volume / 65535.0 + 0.0184)))`
and that works well for me... (the 0.0184 is approx. exp(-100/25) and prevents the result to become negative or undefined)
Thanks again for your support!

maumi, 2016-04-03:
I will try you equation. Looks better than mine. 
But I'm not sure if every sound card behaves identical. 
I will see it when I try it with my usb sound card.

whumann, 2016-04-03:
Well for me the point was that the scale in alsamixer seemed very reasonable already. So I wanted 0..100 in mixer.setvolume() to be the same as 0..100 in alsamixer. But it looks like somewhere in between an exponential is applied because someone expected a linear volume sale in alsamixer. But because alsamixer is already a more-or-less-dB scale, I tried to undo the additional exponential.
What I also don't know:
1. Is the mapping mixer.setvolume() -> alsamixed always the same
2. Is the scale in alsamixer always as good as it is for the DAC+.

maumi, 2016-04-03:
Equation works good for me, thanks.
Only needed to import math.
For me the alsamixer scale is not suitable. With 50% it's nearly silent.
But with the log eq. it works.
Easier then my if-thing.
I think you can close this if your problem is solved.

whumann, 2016-04-03:
Yes, will close...

closed 2016-04-03 by whumann