Fornoth/spotify-connect-web #52: Missing license information

kjagiello, 2016-05-06:
Could you please add a license declaration? I'd love to use some of your code for a hobby project I plan to do during the summer (kind of party jukebox), but I'm not sure if I'm allowed to. Cheers for doing great work!

Fornoth, 2016-07-17:
Sorry for the really late response, but I plan on adding a MIT license to this project https://github.com/Fornoth/spotify-connect-web/issues/63

Also, on the party jukebox thing, you can have more control over the spotify connect protocol if you use librespot
I talked a little bit about a party jukebox thing on gitter: https://gitter.im/sashahilton00/spotify-connect-resources?at=57096e9ec65c9a6f7f28380b

Fornoth, 2016-07-17:
MIT license has now been added in https://github.com/Fornoth/spotify-connect-web/commit/e6e25c48673095e7fabb14d0f05199cba951bc18

closed 2016-07-17 by Fornoth