Fornoth/spotify-connect-web #53: Respect non-default devices when using the --mixer option

vrob01, 2016-05-29:
In console_callbacks.py, line 87 a new mixer instance is created. If the desired mixer, as set by the "--mixer" option is not on found on the default card with cardindex=0, this will critically fail.
If the user chooses a different card via the "--device" option this should be respected here and the mixer should be sought after on the card given by "--device" using the "cardindex" parameter.
Or give the user a mean of selecting the card on which the given mixer is to be found.

cadavre, 2016-06-01:
I have my `--device hw:1,0' and `--mixer PCM`. When I change volume in Spotify no volume is changed for my selected card. Instead – same named mixer for cardindex=0 volume is changed. Specifying device for mixer would be very useful.

Fornoth, 2016-07-05:
Just made a new release with `--playback_device` and `--mixer_device_index` options. 
https://github.com/Fornoth/spotify-connect-web/releases/tag/0.0.3-alpha
`--playback_device` is a replacement for `--device` and uses the names from `aplay -L`
`--mixer_device_index` takes the device index of the card outputting the audio, like the `1` in `hw:1,0`
I'm not sure if there's a way to automatically determine the correct mixer index from the device

exetico, 2016-08-04:
@Fornoth - This is just awesome. I was working with a lot of small hacks, to get it work with Volumio. After reading multiple guides, and nothing gave me the final goal - i just saw this post.

Thanks for making this possible! :-)

Everyting is running fine.

... a note to all people, having troubles finding the right settings/options:

The following command is used in my setup.
`--playback_device default:CARD=DAC --mixer_device_index 5`

Option nr 1:
`--playback_device default`
Run the command: `aplay -L`
Look at the output, to find the right name. I used the option: `default:CARD=DAC`
[image]

Option nr 2:
`--mixer_device_index 5``
Run the command: `alsamixer`
Press F6 to open the overview
[image]
As you can see, here is my USB Audio DAC number 5. Cause of this, the option value is `5`

Hopefully is could be helpfully for someone.

awiouy, 2016-08-12:
@Fornoth 
If the device is specified as hw:a,b, then the mixer index is a.
If the device is specified as default:CARD=card,DEV=d, then the mixer index is alsa.pcms().index(card)
No?

Fornoth, 2016-10-02:
@exetico no problem!
@awiouy Not exactly (for me at least) see http://larsimmisch.github.io/pyalsaaudio/libalsaaudio.html#alsaaudio.mixers

atze09, 2017-02-10:
I have the same problem .
with -- device suni4codec --mixer_device_index 0 i can not change volume

`corected_playback_volume: 100
playback_volume: 49151
corected_playback_volume: 100
playback_volume: 65535
corected_playback_volume: 100
playback_volume: 43908
corected_playback_volume: 100
playback_volume: 5242
corected_playback_volume: 100
`
That says the log.

Fornoth, 2018-01-15:
@atze09 if you're trying to do this on a CHIP, try `--playback_device 'default:CARD=sun4icodec' --mixer 'Power Amplifier'`