Fornoth/spotify-connect-web #55: unable to start spotify-connect-web.sh

pickeld, 2016-06-13:
hello all,
./spotify-connect-web.sh --username name@gmail.com --password password --bitrate 320 -d
while running this ^ line im getting the following error:

>  ESDK: SpCallbackError (at 0xb6f29ee0): error = 8, context = 0xb62e0c86 (underlying Spotify error = -112, underlying OS error = 0

im trying to run spotify-connect-web on raspberry pi 2. 

hope you can help.
thank you
David

closed 2016-06-13 by pickeld

lmachens, 2016-08-18:
I have the same error. What was the reason and how did you solve it?

Fornoth, 2016-10-02:
`error=8` stands for incorrect login https://github.com/Fornoth/spotify-connect-web/blob/master/spotify.h#L17
You need to use your spotify username, not your email