Fornoth/spotify-connect-web #56: Playing on onboard RPI2 instead of Sabre A9023 DAC

kemo87, 2016-06-22:
Hi,
I have RPI2 running latest RuneAudio beta. All working good, if I stream music using RuneAudio, its play from my Sabre A9023 DAC. But if I try Spotify connect its play from onboard RPI2 audio card and DAC is silent. Dont know how to fix it, pls help me and sorry for my really bad english. Thx.

Kemo

mikehorgan, 2016-06-22:
I am using running an RPI3 with an outboard USB DAC. I was able to set the default output to usb by creating the file /etc/asound.conf containing these lines:

```
pcm.!default {
        type hw
        card 1
}

ctl.!default {
        type hw
        card 1
}
```

You can also try adding the parameter **--device 1** to your spotify-connect-web.sh launch command.

This package has been working extremely reliably for me, kudos to the developers.

kemo87, 2016-06-23:
Great! Parametr --device 1 work for me. Thanks for help, and i want thanks a developers for their work. Its sound great on Sabre A9023. Thanks.

Fornoth, 2016-07-03:
I'm gonna close this issue since it seems like you got it working
Feel free to re-open if you need to

closed 2016-07-03 by Fornoth