Fornoth/spotify-connect-web #57: Question: possible to use PulseAudio?

jared-94, 2016-07-03:
Hi,

I would like to install spotify-connect-web on a RaspberryPi and then use VLC to stream the output to my Logitech Media Server.
I can stream with VLC any sound from PulseAudio but not with ALSA.

Do you know a way I can achieve this?
Thanks

Fornoth, 2016-07-05:
Have you tried this? https://forum.videolan.org/viewtopic.php?t=105918
I tried it on my desktop which uses pulseaudio, and played audio with alsa via aplay, and VLC was playing it back just fine

jared-94, 2016-07-05:
Thanks for your anwser.
It works when playing a file with aplay but unfortunately not with spotify-connect-web.
The problem is that it doesn't appear in pavucontrol (a pulseaudio sink manager)

I also tried to install the official spotify app with exagear (which emulate a x86 or x64 ubuntu system on ARM devices), it works but not very stable...