Fornoth/spotify-connect-web #58: Forcing USB audio output

mankowskiwik, 2016-07-04:
Hello everybody, 

I really need your help. I am unable to output audio to my Denon DA 300 USB DAC. Previously while running Volumio 1.55 based on Wheezy I was able to edit alsa-base.conf file and rearrange indexes to that usb sound card would get priority over HDMI and Analog. 

Currently running newest OSMC (Jarvis 16.1) I am unable to force my DAC to be set as output (tried on Volumio 2 as well, same results). According to `aplay -l` output my DAC is card number 1. I have tried: 

- Modify` /etc/asound.conf` and `~/.asoundrc` with following: 

```
pcm.!default {
    type hw
    card 1
}

ctl.!default {
    type hw           
    card 1
}
```

- Modify  `/usr/share/alsa/alsa.conf` setting default pcm and ctl as my card 1

Those solution do not work. In OSMC I can select DAC and it runs perfectly fine. For the rest it does not work. 

Please advise. I am really stuck and have searched a lot, can't really find solution. Thanks! Wiktor

Fornoth, 2016-07-05:
have you tried using `--device` or the new `--playback_device` option to specify your output device?

mankowskiwik, 2016-07-05:
Thanks for the reply. Actually I have managed to solve this issue. Passing -Dplughw:DA300USB parameter solved my problem. However I don't quite understand what is the difference between one I used and --device hw:1 from previous attempts. Is it because its an usb DAC not a GPIO hifiberry and similar? Thanks!

Fornoth, 2016-07-06:
I'm not sure, the `--device` parameter just tries to find the alsa device by guesswork (according to the pyalsaaudio docs)
and you're welcome!

Fornoth, 2016-09-04:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-09-04 by Fornoth