Fornoth/spotify-connect-web #59: Segmentation fault on running spotify-connect-web

TimHausmann, 2016-07-05:
HI, I hope this is the right place for a problem like that. I tried the installation following the intructions but I get a Segmantation fault message when trying to execute `./spotify-connect-web`

closed 2016-07-05 by TimHausmann

Fornoth, 2016-07-05:
I'm guessing you figured out your problem?

TimHausmann, 2016-07-05:
Yeah, I guess it's because I use a Raspberry Pi 1, right? Got it working for me using the latest console_callbacks.py with the web-chroot build. I'm quite a noob to this. Would love to learn more, bit don't want to bother you ;)

Fornoth, 2016-07-06:
Yeah, unfortunately the PyInstaller version isn't available for Rpi 1 for the time being. And feel free to ask me whatever about this!