Fornoth/spotify-connect-web #60: Raspberry Pi 1 Model B

mprahlkamps, 2016-07-05:
Hello! Is there any way to get this to work with Raspberry Pi 1 Model B?
I used [this](https://discourse.osmc.tv/t/howto-setup-a-spotify-connect-web-server-on-a-raspberry-pi-with-osmc/15818) tutorial and then it works, but just for one user.

TimHausmann, 2016-07-06:
it only works for the user with the credentials you used for the spotify-connect-web.sh

mprahlkamps, 2016-07-06:
I've read on some other issues that you can use zeroconf for you to be able to use multiple accounts, right? But that is only available for RaspberryPi 2? If so, is there anything I can do to make the lates version work on RaspberryPi 1?

Thanks for your time!

TimHausmann, 2016-07-06:
I wrote [this guide](https://wordpress.com/stats/post/295/homebrewedthoughtsandthings.wordpress.com) to make it work for me. It uses the latest console_callbacks.py and works fine on a raspberry Pi 1 model B

mprahlkamps, 2016-07-06:
@TimHausmann thanks, but that didn't change anything.
For a single user the chroot works fine. (I don't know how outdated it is?)
But I would like to use this like on the PS4, where multiple (premium)users can connect to a single account on the PS4 and change the music.
Is there a possibility?

Fornoth, 2016-07-06:
Use the command `avahi-publish-service TestConnect _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf` to have multi-user support
https://github.com/Fornoth/spotify-connect-web#logging-in (I'll make that section in the README a little clearer)

mprahlkamps, 2016-07-06:
Sorry for not being clearer. I already used this command. 

I've just opened two ssh windows, on the first I used the command `avahi-publish-service TestConnect _spotify-connect._tcp 4000 VERSION=1.0 CPath=/login/_zeroconf` on the second I stared the server with `./spotify-connect-web.sh --bitrate 320`.

When I now try to connect it doesn't find the Raspberry. Is there anything wrong the way I do it?

Fornoth, 2016-07-07:
Try applying these changes to main.py (in /usr/src/app inside the chroot) https://github.com/Fornoth/spotify-connect-web/pull/16/files

mprahlkamps, 2016-07-07:
Thank you very much! I'ts working now.

Great support :+1:

closed 2016-07-07 by mprahlkamps

Fornoth, 2016-07-07:
You're very welcome!

Rick7221, 2016-10-05:
I have a question about this multi-user support: How does it work with the appkeys? If i want to connect from different accounts, how do i specify the appkey files accordingly?

mprahlkamps, 2016-10-05:
You just need one AppKey. On the main page at the very bottom you'll see how others can login.

mpyzara, 2016-10-13:
If i put spotify-connect-web.sh to autostart should i use avahi command everytime the pi is started?