Fornoth/spotify-connect-web #61: Feature request

shakudu, 2016-07-05:
First, THANK YOU for this amazing piece of software. It made my RPi exactly what i needed it to be :+1: 
My title is vague because i don't really know what i'm asking for.
I'm fiddling around with connecting a 20x4 i2c display to my RPi, i use the Python smbus library to write to it. So what i'm after is a way to write song name/artist, play/pause status, volume and other info to it.
I don't know where to begin.

Fornoth, 2016-07-06:
You're very welcome!
And you should be able to get this stuff from the API (`/api/info/metadata` and `/api/info/status`)
The full URL would be something like `http://localhost:4000/api/info/metadata`

john-a-harris, 2016-07-06:
I'd like to add my thanks too - it's a great solution. I am also looking to do the same as the original poster, but I am using the connect.py direct - is there anyway to get at the meta data this way? Shairport-sync writes its metadata to a pipe - could something similar be done here?

shakudu, 2016-07-06:
It turned out pretty awesome!
[image]
Is there a way to output what device Spotify is playing from in the metadata? It changes even if i play on my cellphone or PC :)

Fornoth, 2016-07-07:
@john-a-harris I'll look into doing something like that
@shakudu Unfortunately not :( That info isn't exposed in the library I'm using

john-a-harris, 2016-07-12:
@shakudu Looks great! Are you polling the api, or using some other trigger for deciding when to fetch the data?

shakudu, 2016-07-13:
Polling the api every few seconds and outputs to display if there's any changes.
But some kind of trigger would've been more effective.

michaellunzer, 2017-08-23:
hi @shakudu do you mind posting your code or explaining a bit more of how you accomplished it? I'm hoping to do a similar thing with the same LCD display.