Fornoth/spotify-connect-web #62: mixer.getvolume()[0] is negative

willcode4food, 2016-07-10:
I'm trying to get this to run on one of those new fancy [C.H.I.P.](https://nextthing.co/pages/chip) computers .  And I am getting an error when running the chroot version 

  File "main.py", line 27, in <module>
    connect_app = Connect(web_error_callback)
  File "/usr/src/app/connect.py", line 75, in __init__
    lib.SpPlaybackUpdateVolume(mixer_volume)
OverflowError: can't convert negative number to unsigned

I can play audio files using aplayer specifying device hw:0,0 without issues.  For some reason, the mixer.getvolume() call is returning a negative value.  alsamixer allows me to toggle the volume without issue.   Any help would be appreciated.

Fornoth, 2016-07-10:
If you're using the built in audio hardware, try passing `--playback_device 'default:CARD=sun4icodec' --mixer 'Power Amplifier'` as options

willcode4food, 2016-07-11:
That worked!  Thanks!

closed 2016-07-11 by willcode4food

fightforlife, 2016-11-28:
I am sorry for reopening this, but I can not use the --mixer option because "Power Amplifier" has a space in it.

./spotify-connect-web.sh ...  --bitrate 320 --playback_device 'default:CARD=sun4icodec' --mixer 'Power Amplifier'

File "/usr/src/app/console_callbacks.py", line 28, in <module>
    mixer = alsa.Mixer(args.mixer)
alsaaudio.ALSAAudioError: Unable to find mixer control 'Power',0

Fornoth, 2016-12-12:
@fightforlife try putting double quotes around `--mixer` itself, like `"--mixer 'Power Amplifier'"`