Fornoth/spotify-connect-web #64: ALSAAudioError: Data size must be a multiple of framesize

naguirre, 2016-07-20:
When playing with an USB audio carte (USB-Audio - Combo384 Amanero) i get the following traceback : 

```
kSpPlaybackNotifyPlay
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/src/app/console_callbacks.py", line 107, in playback_thread
    device.write(data)
ALSAAudioError: Data size must be a multiple of framesize 
```
Any idea ?

Fornoth, 2016-07-23:
What device are you running this on?

Fornoth, 2016-09-04:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-09-04 by Fornoth

redvoodoochild, 2016-11-21:
Hi.
Excuse my poor english.
It is possible to re-open this issue ? I've got the same problem.
I'm trying to run spotify-web-connect on Libreelec with my usb DAC.

```
LibreELEC:~/.kodi/addons/service.spotify-connect-web# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
default:CARD=DA300USB
    DA-300USB, USB Audio
    Default Audio Device
sysdefault:CARD=DA300USB
    DA-300USB, USB Audio
    Default Audio Device
front:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    Front speakers
surround21:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=DA300USB,DEV=0
    DA-300USB, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
```
```
LibreELEC:~/.kodi/addons/service.spotify-connect-web # ./spotify-connect-web -n Salon -o iec958:CARD=DA300U
SB,DEV=0
Loading Spotify library...
min_volume_range: 0.0
Using libspotify_embedded version: release-esdk-1.20.0-v1.20.0-g594175d4
SpInit: 0
playback_volume: 60947
corected_playback_volume: 92
public key: xxxxxxxxxxxxx
device id: 0970b14f-86f5-4f55-983e-3590f68d4ffa
remote name: Salon
account req: PREMIUM
device type: AUDIODONGLE
kSpConnectionNotifyLoggedIn
iXFWVkc42sVam1QsiOwJcpjBpgSjtoI4QvujQJQJng2kRmja9fTCS5d0ZTpje/FS
kSpPlaybackNotifyPause
kSpPlaybackNotifyShuffleEnabled
playback_seek: 56863
kSpPlaybackNotifyTrackChanged
kSpPlaybackNotifyBecameActive
playback_seek: 56863
kSpPlaybackEventAudioFlush
kSpPlaybackNotifyPlay
AlsaSink: device acquired
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
Ups! Some badness happened:  Data size must be a multiple of framesize
kSpPlaybackNotifyPause
AlsaSink: device released
^CkSpPlaybackEventAudioFlush
kSpConnectionNotifyLoggedOut

```
There's no sound.
My DAC works very well with mpd

Thnak you for your help.