Fornoth/spotify-connect-web #65: Help with RPi Zero

ashthespy, 2016-07-22:
I just completed building from source, and run into the following when I try and run `main.py`. Any guidance would be appreciated!

`
volumio@volumio:~/spotify-connect-web$ python connect.py -u User-p Pass-d
Loading Spotify library...
Traceback (most recent call last):
  File "connect.py", line 8, in <module>
    from connect_ffi import ffi, lib, C
  File "/home/volumio/spotify-connect-web/connect_ffi.py", line 23, in <module>
    libraries=[str('spotify_embedded_shared')])
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 437, in verify
    lib = self.verifier.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 114, in load_library
    return self._load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 225, in _load_library
    return self._vengine.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 158, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/home/volumio/spotify-connect-web/__pycache__/_cffi__x4c785defx96dcd04b.arm-linux-gnueabihf.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory
volumio@volumio:~/spotify-connect-web$ ls
=0.9.2           README.md       connect_ffi.pyc                requirements.txt     spotify_appkey.key
Dockerfile       __pycache__     console_callbacks.py           run-with-docker      static
LICENSE          connect.py      libspotify_embedded_shared.so  spotify.h            templates
Makefile.docker  connect_ffi.py  main.py                        spotify.processed.h  utils.py
`

ashthespy, 2016-07-22:
Update:
Everything seems to work five with a pre-built chroot. But when I compile from source to get the recent mixer changes, I still run across these errors.

I have placed the `libspotify_embedded_shared.so` into `/urs/lib` as well as directly into the directory of the python scripts. No dice though, the script doesn't seem to be able to find the library.

Any pointers? I feel like I am missing something extremely trivial?

````
volumio@volumio:~/spotify-connect-web$ LD_LIBRARY_PATH=$PWD python main.py
Loading Spotify library...
Traceback (most recent call last):
  File "main.py", line 14, in <module>
    from connect_ffi import ffi, lib
  File "/home/volumio/spotify-connect-web/connect_ffi.py", line 23, in <module>
    libraries=[str('spotify_embedded_shared')])
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 437, in verify
    lib = self.verifier.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 114, in load_library
    return self._load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/verifier.py", line 225, in _load_library
    return self._vengine.load_library()
  File "/usr/local/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 158, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/home/volumio/spotify-connect-web/__pycache__/_cffi__x4c785defx96dcd04b.arm-linux-gnueabihf.so': libspotify_embedded_shared.so: cannot open shared object file: No such file or directory
````

Fornoth, 2016-07-23:
Fixed on gitter: https://gitter.im/sashahilton00/spotify-connect-resources?at=5792a61b959ee82b3e3e12e0

closed 2016-07-23 by Fornoth