Fornoth/spotify-connect-web #68: Nginx proxy_pass

joeinnes, 2016-08-01:
Hi all,

I'm having a few problems setting up a reverse proxy for this.

I have a statically hosted page on my RPi which is handling my media centre, and I want to add a link to it to the spotify-connect-web HTML UI.

I tried with a normal `proxy_pass`, which didn't work (404), and then tried to use upstreaming, but that didn't work either (404 too).

When I ssh into my Pi and wget localhost:4000, it downloads the raw HTML for the web interface, so I'm not sure what's going wrong here.

```
upstream spotifyweb {
  server localhost:4000;
}

server {
  // snip
  location /spotifyweb/ {
    proxy_pass http://spotifyweb;
  }
  // snip
}
```

Any ideas about what could be wrong? I'm a little confused as to why it's 404-ing instead of 5XX-ing, which suggests maybe it's something to do with my nginx config rather than spotify-connect-web.

Best regards,

Joe

joeinnes, 2016-08-01:
OK, getting there... added a trailing '/' to the proxy_pass directive, but now none of the resources are found...

Fornoth, 2016-09-04:
The resources are served from an absolute path, so there's two options:
- Tell Flask what the url path for the static folder should be by adding the `static_url_path` parameter to this line: https://github.com/Fornoth/spotify-connect-web/blob/master/main.py#L39
- Map the `/static` path in Nginx to the proper path