Fornoth/spotify-connect-web #69: Can you add the --cors option?

CatoAntonsen, 2016-08-05:
I try to use the CORS argument that main.py offers, but it doesn't seem that spotify-connect-web has implemented it...?

Anyway, thanks for a GREAT and easy way of making my RPi a Spotify Connect'ed device! :-)

exetico, 2016-08-23:
Nope :-( That's SAD. Damn..

Fornoth, 2016-09-04:
Even though it doesn't show it in `--help`, the option should still work (at least for 0.0.3)
Try doing `grep -i cors main.py` in the main directory

exetico, 2016-09-04:
Well Fornoth - As you can see, the option isn't supported. But, maybe we have to use it, in another way.

Fornoth, 2016-09-05:
@exetico can you double check that you're running the 0.0.3 release by checking if main.py has any CORS references?

exetico, 2016-09-05:
I can see that "CORS" is in the top header as "import CORS".
I'm using the Alpha-build of 0.0.3 currently. So.. Maybe that why?
https://github.com/Fornoth/spotify-connect-web/releases/download/0.0.3-alpha/spotify-connect-web_0.0.3-alpha.tar.gz

Fornoth, 2016-09-05:
@exetico can you try re-downloading the 0.0.3 release? I've tried it on two devices, and the --cors argument exists

Fornoth, 2016-11-06:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-11-06 by Fornoth