Fornoth/spotify-connect-web #71: pre-built chroot out of date?

TheMattyBoy, 2016-08-08:
I followed instructions to add the pre built chroot, but the chroot seems to have an old version of the code. e,g, the new command line options such as playback_device , mixer_device_index are not accepted when starting ./spotify-connect-web.sh 

Is the chroot being built with the latest release? Doesn't seem so... I compared the main.py code from chroot and version 0.0.3 Alpha and looks like main,py in the chroot version is from way back.

Bit of a noob on this so forgive me if this is a stupid question.

axelsimon, 2016-08-18:
Just had a quick look, it looks indeed like the version in the pre-built chroot is out of date, as it doesn't have that many options (it still has the deprecated --device option instead of --playback_device for instance).

Furthermore, taking about the pre-built chroot, it could be a good idea to slim it down, by removing a lot of the documentation that is bundled with python, and /usr/share etc. I let @Fornoth judge of that however ;)

Fornoth, 2016-09-04:
Yeah, it's outdated unfortunately, I'll try and get it updated soon.
And I do plan on offering a slimmed down version once I get the build scripts for Jenkins done

Fornoth, 2016-09-05:
I uploaded a new chroot based off of the 0.0.3 release https://github.com/Fornoth/spotify-connect-web/releases/tag/0.0.3-alpha

Fornoth, 2016-10-02:
I'm gonna close this for now, feel free to re-open if you're still having problems

closed 2016-10-02 by Fornoth