Fornoth/spotify-connect-web #72: last.fm scrobbling

cadavre, 2016-08-09:
Is it possible to add last.fm scrobbling? I always used last.fm to make statistics for my music and since I'm using only my RPi with spotify-connect I have no archive of playbacks. :/

guilro, 2016-08-20:
I would love to contribute this feature, I plan to do it next week !

cadavre, 2016-08-23:
You inspired me to work. :p I managed to enable scrobbling, but I'm new to Python so my code probably looks like a s**t.

Basically we need: LastFM API key and secret, username and password

How it works: on `kSpPlaybackNotifyTrackChanged` I'm starting Timer with 30s (time to play to scrobble) callback for `lastfm.scrobble()` where artist and track title is provided by `utils.get_metadata()`. My solution uses `pylast`. I also included simple checks for `isPlaying` so it's not scrobbling a track this is not playing and canceling `Timer` on track change.

cadavre, 2016-08-23:
I'll try to cleanup my code, refactor params (api keys, username etc.), add check if all Last.fm params are given (if not - disable scrobbling) and few more things. Then I'll PR so enable further work.

guilro, 2016-08-24:
I did almost the same this weekend, but did not PR anything because my last
FM id was still hard coded. For the timer, I listen on play and pause event
to respect what last FM says in their doc about scrobbling : a song should
be scrobbled when it has been played for half of its duration or for at
least 4 minutes. I also implemented Now Playing.

I'll try to push my code today so we can work together !

guilro, 2016-08-24:
You can find my work in this commit :
https://github.com/guilro/spotify-connect-web/commit/37f9741a1f2c45e2e2758ad49f97df93fd115aa6

missing :
- [ ] configuration of Last.fm API credentials
- [ ] optional use of Last.fm

guilro, 2016-08-26:
Thanks a lot for your comments on my first commits :)

I made a PR here : https://github.com/Fornoth/spotify-connect-web/pull/83

Please don't hesitate to comment again or to publish your code to see if we can choose and improve one of the implementations.

Fornoth, 2016-12-12:
this has been added via #83

closed 2016-12-12 by Fornoth