Fornoth/spotify-connect-web #74: packaged release: symbol lookup error

TheMattyBoy, 2016-08-11:
Have installed the packaged release 0.0.3 Alpha, but am getting the following errors when running ./spotify-connect-web in the extracted directory

./spotify-connect-web --name Rune --username XXXX --password YYYY--bitrate 320
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
Loading Spotify library...
min_volume_range: 0.0
Using libspotify_embedded version: release-esdk-1.20.0-v1.20.0-g594175d4
SpInit: 0
playback_volume: 0
Mute activated
public key:  <<redacted>>
device id: <<redacted>>
remote name: Rune
account req: PREMIUM
device type: AUDIODONGLE

The web interface runs, but reports as not being logged in, and the device doesn't show in the list of available devices in Spotify.

Any ideas on how to fix?- I'm running it on a Rasperyy Pi 2 that has Rune Audio installed (Arch Linux)

Thanks in advance,

Matt

danifr, 2016-08-11:
I remember seeing those error messages too, although my issue was a little bit different.
I ended up installing the package directly from AUR https://aur.archlinux.org/packages/spotify-connect-web/

TheMattyBoy, 2016-08-11:
Thanks danfr, appreciate your suggestion. I downloaded the tarball from AUR, but I'm getting error msgs when trying to build the package

makepkg -Acs                
==> Making package: spotify-connect-web 0.0.3.alpha.r6.g565767b-1 (Thu Aug 11 15:47:30 CEST 2016)
==> Checking runtime dependencies...
==> Installing missing dependencies...
error: target not found: python2-flask-bootstrap
error: target not found: python2-flask-cors
error: target not found: python2-pyalsaaudio
==> ERROR: 'pacman' failed to install missing dependencies.

I'm a linux noob so this has me stumped - does this mean the tarball is listing dependencies that are incorrect?

Thanks for any help you can provide. Really want to get playing spotify on my PI, but just keep hitting issues.. ;-(

Cheers

danifr, 2016-08-11:
Yeah, I forgot I had to deal with this... my bad, sorry.
Apparently there is not python-flask-bootstrap package for python2 anymore.
I will add this package to AUR so you can install it from there. I'll try to do it today.

You can find the other 2 packages (python2-flask-cors and python2-pyalsaaudio) in AUR already.
Just install them from there or automate their installation using yaourt.

danifr, 2016-08-11:
Hi again,
I just submitted python2-flask-bootstrap to AUR (https://aur.archlinux.org/packages/python2-flask-bootstrap)
Can you try to install it again?

TheMattyBoy, 2016-08-12:
H danifr, and thanks so much for going to all this effort to help. With your help and some perseverance I did manage to get the build to complete.

My findings were that:

1. The  new python2-flask-bootstrap required two additional packages 'python2-dominate' and 'python2-visitor'. Once these were installed I could build python2-flask-bootstrap.
 
2. The 'python-pyalsaaudio' package wouldn't build on my PI, I had to change the PKGBUILD line 
 arch=('i686','x86_64')
 to  
arch=('any')

With those changes I was able to build and install the AUR package [https://aur.archlinux.org/packages/spotify-connect-web/](https://aur.archlinux.org/packages/spotify-connect-web/)

Time for bed now, will report if I get any further. I'm a windows dev, so have had to dip my toe into Linux far more than I thought tonight!

TheMattyBoy, 2016-08-12:
Hi Danifr - Just an update to say that I now have Spotify connect working great on my PI.  With the newer version (0.0.3) spotify connect works well in conjunction with Rune Audio (spotify connect releases audio when not playing), and can now also get the volume control to work by using the new argument mixer_device_index. Happy days!

Thanks gain for your help, it's appreciated.

closed 2016-08-12 by TheMattyBoy