Fornoth/spotify-connect-web #75: webserver shuffle/repeat: Internal Server Error

noemu, 2016-08-12:
Hi,
I'm using the spotify-connect-web_0.0.3-alpha RPI Release.

The Webserver Shuffle and Repeat function doesn't work.

GET <ip:4000>/api/playback/shuffle  gives an "500 (INTERNAL SERVER ERROR)"
same with repeat.

Everything Else (play/pause/prev/next) and (/api/info/metadata, /api/info/status) works perfectly.

Fornoth, 2016-11-07:
This is fixed in the source (7b1c426), but I haven't made a release with it yet

Alootje, 2017-02-13:
Hello Fornoth,

Is there an ETA when you think this will be released?

nooryani84, 2017-02-13:
@Alootje https://github.com/Fornoth/spotify-connect-web has build instructions if you want to experiment yourself.

Fornoth, 2018-01-15:
A new build with the fix has been released 
https://github.com/Fornoth/spotify-connect-web/releases/tag/0.0.4-alpha

closed 2018-01-15 by Fornoth