Fornoth/spotify-connect-web #76: ALSAAudioError on Runeaudio

ThomasBorghs, 2016-08-13:
I get following error using the latest package released version: 0.0.3-alpha on runeaudio 0.3-beta. I'm trying to output to a USB dac (details below). As you can see in the command I tried passing as much parameters as possible, it keeps giving the same error with any other combination of these parameters, including when I don't pass any parameters other then bitrate and name.

[knuppel@runeaudio spotify-connect-web]$ ./spotify-connect-web --bitrate 320 --name homeserver --device M2496 --playback_device M2496 --mixer M2496 --mixer_device_index 1
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
sh: symbol lookup error: sh: undefined symbol: rl_executing_keyseq
Loading Spotify library...
Traceback (most recent call last):
  File "loader.py", line 3, in <module>
  File "runpy.py", line 180, in run_module
  File "runpy.py", line 72, in _run_code
  File "/home/knuppel/spotify-connect-web/main.py", line 14, in <module>
    from connect import Connect
  File "/home/knuppel/spotify-connect-web/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/home/knuppel/spotify-connect-web/console_callbacks.py", line 24, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
alsaaudio.ALSAAudioError: No such file or directory [default]
Failed to execute script loader 

[knuppel@runeaudio spotify-connect-web]$ sudo aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: M2496 [Musical Fidelity V90-DAC 24/96], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

The chroot version works perfectly with the DAC as well, but I want to run multiple instances, 1 for my account, 1 for my gf's (which worked on osmc when I tried it). I tried fiddling around with the naming of "default" parameter assignments, but wasn't able to figure out where the value came from or which call causes the error (my Python knowledge is quite lacking). I would think the problem is linked to the arch linux distribution lacking some package perhaps? alsamixer and amixer commands are available though, so I wasn't sure if anything else was needed.

Any help would be appreciated!

TheMattyBoy, 2016-08-19:
Hi - I couldn't get the 0.0.3 Alpha to work on Rune Audio either. Think you're right that there's some kind of issue relating to Arch. To get 0.0.3 to run, I had to install the following packages from AUR

python2-flask-bootstrap.tar.gz
python2-flask-cors.tar.gz
python-dominate.tar.gz
python-pyalsaaudio.tar.gz  (note you'll need to edit the arch version in the PKGBUILD file to 'any' to get this to build on the PI)
python-visitor.tar.gz
spotify-connect-web.tar.gz

That did the trick, and now I have the 0.0.3 Alpha working great on my Rune Audio.

Would be much simpler for everyone if the chroot was updated to the newest code version though!

Hope that helps.

danifr, 2016-08-19:
Hi,

>  python-pyalsaaudio.tar.gz (note you'll need to edit the arch version in the PKGBUILD file to 'any' to get this to build on the PI)

according to the maintainer of the package you just need to build it using  `--ignorearch`.

>[carstene1ns commented on 2016-03-20 18:42](https://aur.archlinux.org/packages/python-pyalsaaudio/)

>@bjo: This builds a C extension, so 'any' architecture is wrong. I have read somewhere that arm users build using `makepkg --ignorearch` (or -A), is that not true anymore? If so, I can add the additional ones.

It worked for me.

Also, I had the same error you described (https://gitter.im/sashahilton00/spotify-connect-resources?at=5793cd65959ee82b3e47a144) on my Pi2 running Arch. As @TheMattyBoy said already, it can be worked around installing the package from AUR.

Fornoth, 2016-09-04:
The `sh: symbol lookup error` should be harmless, but the error involving `alsa.mixers()[0]` is probably a hardware problem. I'm gonna work on adding a better error message for that

fosple, 2016-11-09:
Same here...
`./spotify-connect-web.sh --username XXXX --password XXXX --bitrate 320 -o sysdefault:CARD=Device -m PCM`

results in

`Loading Spotify library...
Traceback (most recent call last):
  File "main.py", line 14, in <module>
    from connect import Connect
  File "/usr/src/app/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/usr/src/app/console_callbacks.py", line 24, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
alsaaudio.ALSAAudioError: No such file or directory [default]
`

I'm using a USB Soundcard, which is working fine with e.g. mplayer. Not sure if that might cause the problem...

`aplay -L
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
equal
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions`

Fornoth, 2016-12-12:
@fosple for the `-m` option, check the output of `amixer` instead to get the proper device name