Fornoth/spotify-connect-web #77: Default decibel range off by a factor of 100

markubiak, 2016-08-15:
Using 0.0.3 on a Raspberry Pi 2 to an external DAC.  The --dbrange setting only worked with a value of 1.  Upon inspection of the commit that added --dbrange, I changed the line
`+volume_range = (mixer.getrange()[1]-mixer.getrange()[0]) / 100` to `+volume_range = (mixer.getrange()[1]-mixer.getrange()[0])` which resolved the issue.

cadavre, 2016-08-22:
Confirmed. Removing `/ 100` allowed me to set --dbrange 53 for my DAC.

Nosskirneh, 2016-08-24:
I also have the issue that the range off is completely off.

When removing `/ 100` I now get that all steps are 99 (besides the two end-points). I've tried different options for --dbrange but in my opinion that setting lacks explanation.

Like this:
```
corected_playback_volume: 100
playback_volume: 64879
corected_playback_volume: 99
playback_volume: 64224
corected_playback_volume: 99
playback_volume: 61602
corected_playback_volume: 99
playback_volume: 58981
corected_playback_volume: 99
playback_volume: 55704
corected_playback_volume: 99
playback_volume: 51772
corected_playback_volume: 99
playback_volume: 49151
corected_playback_volume: 99
playback_volume: 46529
corected_playback_volume: 99
playback_volume: 45219
corected_playback_volume: 99
playback_volume: 42597
corected_playback_volume: 99
playback_volume: 39321
corected_playback_volume: 99
playback_volume: 37354
corected_playback_volume: 99
playback_volume: 35388
corected_playback_volume: 99
playback_volume: 34733
corected_playback_volume: 99
playback_volume: 32112
corected_playback_volume: 99
playback_volume: 29490
corected_playback_volume: 99
playback_volume: 28180
corected_playback_volume: 99
playback_volume: 24903
corected_playback_volume: 99
playback_volume: 21626
corected_playback_volume: 99
playback_volume: 18349
corected_playback_volume: 99
playback_volume: 12451
corected_playback_volume: 99
playback_volume: 3932
corected_playback_volume: 99
playback_volume: 0
Mute activated
```

I'm using amixer to enable AUX audio output:
`amixer cset numid=3 1`

```
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
  : values=1
```

and from searching online I figure that the `min=0` and `max=2` seems weird. Could this be the issue?

Using OpenELEC (Rasplex) with _spotify-connect-web_ 0.0.3.

awiouy, 2016-08-24:
amixer cset numid=3 1 sets the playback route (0 is autodetect, 1 is jack, 2 is HDMI).

To set the volume use amixer set PCM 100%

Nosskirneh, 2016-08-24:
Thank you for answering. Maybe I misunderstood OP's first problem, thought maybe it could've been the same since I thought it had to do with `--dbrange`. Yes, I know what the `amixer` command does - what I'm trying to do is to make the volume scale to be more linear. Right now it's like this:
0&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;0&nbsp;&nbsp;&nbsp;&nbsp;20&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;100
████████████████████

or with `--dbrange 50` (or any other int for that matter):
0&nbsp;&nbsp;&nbsp;&nbsp;100&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;100
████████████████████

What I however want:
0 &nbsp;20&nbsp;30&nbsp;40&nbsp; 50 &nbsp;60&nbsp; 70&nbsp;80&nbsp;90&nbsp; 100
████████████████████

Nosskirneh, 2016-08-25:
Nevermind, I fixed it.

For anyone else having the same issue, this was that I did:
```
RasPlex:~ # amixer controls
numid=3,iface=MIXER,name='PCM Playback Route'
numid=2,iface=MIXER,name='PCM Playback Switch'
numid=1,iface=MIXER,name='PCM Playback Volume'
numid=5,iface=PCM,name='IEC958 Playback Con Mask'
numid=4,iface=PCM,name='IEC958 Playback Default'
```

using numid=1 instead of numid=3 did give better values on `min `and `max`.
```
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=1
  | dBscale-min=-102.39dB,step=0.01dB,mute=1
```

`--dbrange 25` turned out to be most fitting audio scale.

karlvr, 2017-02-02:
I _think_ the right fix to the dbrange issue is to scale `selected_volume_range` rather than remove the scale on `volume_range`? I have made a PR for it #116

karlvr, 2017-02-02:
Apologies, there is of course no difference between scaling one or the other! I think it is valuable to use floating point division though: updated the PR.