Fornoth/spotify-connect-web #79: Segmentation Fault on RPI

l3d00m, 2016-08-20:
If I try to run this util, I get a ´segmentation fault´. I'm using the newest 0.3 version on a Raspberry Pi 1. 

I've put my appkey in place, and commented the line with zeroconf_var in connect.py.

Thanks for your help!

Geophagus, 2016-09-01:
Hey,

I had the same problem but used this: https://github.com/JozoVilcek/spotify-connect-web and it seems to work.

Fornoth, 2016-09-04:
The packaged release isn't compatible with RPi 1 for now, you'll have to build spotify-connect-web from source

flori4n, 2016-09-15:
I did have this issue with another git project. 
Try to create a swap file [http://www.cyberciti.biz/faq/linux-add-a-swap-file-howto/](swap-file-howto) and let us know

Fornoth, 2016-10-02:
Also, you have the option of installing from chroot now since I updated it to be on par with the chroot

h0lger, 2016-10-19:
Is it possible to compile for RPi 1? I would like to use the playback_device you have implemented. The source only have the device option.

Fornoth, 2016-11-07:
@h0lger use the chroot option, it works with RPi 1, and should have the same features as the packaged release

l3d00m, 2016-11-11:
Thanks for your help!

You may close this now.

l3d00m, 2016-11-14:
Maybe this could help you @NewBe001 ?  
http://stackoverflow.com/questions/36879168/error-installing-composer-curl-23-failed-writing-body-0-16133

closed 2021-02-02 by l3d00m