Fornoth/spotify-connect-web #82: Installing with OpenELEC

Nosskirneh, 2016-08-23:
Tried both chrooted and not. OpenELEC is a linux read-only filesystems for media centers. `/storage` is however writeable. Python is preinstalled, but pip is not. 

Maybe this belongs more to the github repo of OpenELEC, but I'll give this a shot since I don't know what's wrong.

```
wget https://github.com/Fornoth/spotify-connect-web/releases/download/0.0.3-alpha/spotify-connect-web_0.0.3-alpha.tar.gz
tar zxvf spotify-connect-web_0.0.3-alpha.tar.gz
```

and then running it `./spotify-connect-web` in the extracted dir outputs this:

```
Loading Spotify library...
Traceback (most recent call last):
  File "loader.py", line 3, in <module>
  File "runpy.py", line 180, in run_module
  File "runpy.py", line 72, in _run_code
  File "/storage/spotify-connect-web/main.py", line 14, in <module>
    from connect import Connect
  File "/storage/spotify-connect-web/connect.py", line 9, in <module>
    from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
  File "/storage/spotify-connect-web/console_callbacks.py", line 19, in <module>
    audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
alsaaudio.ALSAAudioError: No such file or directory [default]
Failed to execute script loader
```

Is alsaaudio missing? Do I have to install it with pip? Tried installing pip but that doesn't work. Suggestions?

cadavre, 2016-08-23:
What version of alsaaudio do you have installed? (Check in requirements.txt)

Try this https://github.com/Fornoth/spotify-connect-web/pull/80

awiouy, 2016-08-23:
This might help: https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages/addons/service/spotify-connect-web

Nosskirneh, 2016-08-24:
Thank you! This fixed it.

closed 2016-08-24 by Nosskirneh

gLarsVegas, 2016-10-31:
@Nosskirneh How did you get spotify-connect installed properly on Rasplex? Still struggling with the same error message. Deleting libasound.so.2 didnt fix it for me.

Nosskirneh, 2016-11-01:
@gLarsVegas I don't remember exactly what I did. [I did fork it though](https://github.com/Nosskirneh/spotify-connect-web) and uploaded what's working for me as we speak. You could try cloning that.

gLarsVegas, 2016-11-01:
Thanks. Havent seen your fork. Passing a card id to alsa.mixers() in console_callback.py fixed it for me.