Fornoth/spotify-connect-web #84: Volume control

Arthur-min, 2016-09-10:
Hello,

I installed spotify connect web, it works, but I'm not able to use volume controls from my phone.
When I use volume buttons, playback_volume and corected_playback_volume are well updated, but have no effect.

> $ ./spotify-connect-web.sh --username woop --password woop --bitrate 320 --name Pi3 --device 1

I tried with -m PCM, same problem.

> $ amixer
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback -982 [87%] [-9.82dB] [on]

What can I do ? Thanks !

closed 2016-09-10 by Arthur-min

binderth, 2016-09-28:
Hi @Arthur-min,

I've got the same issue, my USB-soundcard won't react to volume changes from the spotify client. Where do you set the "--Playback_device hw:1" part? 

Thanks for your help,
Thomas.

Fornoth, 2016-10-02:
@binderth the `--playback_device hw:1` is set the same place as `--username`. Use `amixer` to find the correct device

binderth, 2016-10-04:
oh. I think, I don't have the latest nightly build on my pi. The --Playback_device argument isn't supported by my version. I'll try to git-pull the latest Version then.

Thanks for the help, @Arthur-min!